ReFirmLabs/binwalk #98: External extractor return value not handled correctly

ddcc, 2015-01-07:
Currently, the rule execution function execute() in extractor.py assumes that all non-zero return values are errors. However, this is not the case for all extractors, e.g. for gzip: exit status is normally 0; if an error occurs, exit status is 1; if a warning occurs, exit status is 2. In contrast, for e.g. 7z: 0 = No error, 1 = Warning (Non fatal error(s)), 2 = Fatal error, 7 = Command line error, 8 = Not enough memory for operation, 255 = User stopped the process.

This causes problems with certain gzipped files that may contain trailing garbage; executing gzip -d -f %e will return "gzip: rootfs.gz: decompression OK, trailing garbage ignored" with retval 2, which should be considered a success. But currently, this will be interpreted as a failure and trigger execution of the next extractor, the internal GzipExtractPlugin.

Unfortunately, I don't see an easy way to solve this problem. Perhaps each rule could be annotated with a mapping between return values and errors, or alternatively the meaning of return values could be hardcoded into the execute() function.

devttys0, 2015-01-15:
Adding an option to list return codes that should be considered successes would probably be the best course of action. I think this could be added to the extraction rule syntax without too much difficulty. For any extraction rules that don't explicitly specify a list of success codes, 0 will be assumed.

devttys0, 2015-01-15:
I've checked in an update that supports specifying a comma separated list of successful return codes for each extractor. The `extract.conf` rules have been updated for `gzip` and `7z` appropriately, but other rules may also need to be updated as well.

closed 2015-01-15 by devttys0