ReFirmLabs/binwalk #99: Error when attempting to read firmware

overflowingInt, 2015-01-14:
Getting an error when I attempt to open two other firmwares. Binwalk v2.1.0

Just to be sure, I grabbed one off a forum post to double check but I still receive the error below.  (http://www.devttys0.com/2014/02/reversing-the-wrt120n-firmware-obfuscation/)

 $ binwalk FW_WRT120N_1.0.07.002_US.bin 

Signature Exception: Failed to convert value 'blocksize:' to an integer on line '>>12    belong          blocksize: %d bytes,'
----------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 496, in main
    self.init()
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/signature.py", line 103, in init
    self.magic.load(f)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/magic.py", line 744, in load
    self.parse(lines)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/magic.py", line 764, in parse
    sigline = SignatureLine(line)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/magic.py", line 179, in __init__
    raise ParserException("Failed to convert value '%s' to an integer on line '%s'" % (self.value, line))
ParserException: Failed to convert value 'blocksize:' to an integer on line '>>12    belong          blocksize: %d bytes,'
----------------------------------------------------------------------------------------------------

devttys0, 2015-01-15:
This error indicates that there is a magic signature that could not be parsed; however, the signature line that is reportedly failing does not exist in any of the current binwalk signature files.

This is likely a left over signature from a previous installation; try uninstall, then re-installing binwalk:

```
$ sudo python setup.py uninstall
$ sudo python setup.py install
```

devttys0, 2015-01-27:
Did this resolve you issue?

overflowingInt, 2015-01-27:
Sorry been on travel, looks like if I uninstall and reinstall it worked. Thanks for the reply!

closed 2015-01-27 by overflowingInt