ReFirmLabs/binwalk #103: Entropy functionality depends on pyqtgraph

marksteward, 2015-01-21:
So checking for pyqtgraph before parsing the file will save some wasted time.

devttys0, 2015-01-21:
Binwalk only attempts to load pyqtgraph if it is trying to generate a visual graph. This way, even if you can't, or don't want to, install pyqtgraph, you can still get useful textual information from binwalk's output regarding file entropy.

Specifying `--nplot` explicitly tells binwalk not to do the graphing, so this will suppress any loading errors.

marksteward, 2015-01-21:
Yep, but it would be nice for it to fail to load pyqtgraph (if it's going to) before parsing the file. Then someone can explicitly pass --nplot the first time round instead of the second. Admittedly only a small thing, but also only a small change.

Happy to send a PR tomorrow - I've raised this ticket only because I wanted something to refer back to.

devttys0, 2015-01-25:
Added explicit check for `pyqtgraph` at the beginning of entropy scans. Prints a warning and assumes `--nplot` if the `pyqtgraph` import fails.

closed 2015-01-27 by devttys0