ReFirmLabs/binwalk #109: JFFS2 extraction with truncated data

sviehb, 2015-02-13:
There is a problem with JFFS2 extraction eg. with this firmware: ftp://ftp2.zyxel.com/EMG2306-R10A/firmware/EMG2306-R10A_V1.00(AAJM.3)C0.zip
Files with size >=0x1000 have just null-bytes after 0x1000. This problem occurs when mounting the image directly, so `mtd-utlis` (`jffs2reader`) is not to blame I guess.

JFFS2 in this firmware referenced above uses `zlib`-compression. `uncompress`works for the first inode (file `openssl.cnf`, csize=1768, #ino 442). After that something weird goes on with the next inode(s) which should also belong to `openssl.cnf`.

```
         Dirent     node at 0x001c57dc, totlen 0x00000033, #pino   160, version   440, #ino       442, nsize       11, name openssl.cnf
         Inode      node at 0x001c5810, totlen 0x0000072c, #ino    442, version     1, isize     9374, csize     1768, dsize     4096, offset        0
         Inode      node at 0x001c5f3c, totlen 0x000006a3, #ino    442, version     2, isize     9374, csize     1631, dsize     4096, offset     4096
Wrong data_crc at  0x001c5f3c, 0x80784404 instead of 0xe1b89de0
Empty space found from 0x001c65e0 to 0x001c7000
Empty space found from 0x001c700c to 0x001c8000
Wrong bitmask  at  0x001c8000, 0x016c
         Dirent     node at 0x001c8238, totlen 0x0000003b, #pino   165, version   445, #ino       447, nsize       19, name zcfg_defaultStr.bin
         Inode      node at 0x001c8274, totlen 0x0000007d, #ino    447, version     1, isize       57, csize       57, dsize       57, offset        0
         Dirent     node at 0x001c82f4, totlen 0x00000038, #pino   165, version   446, #ino       448, nsize       16, name zcfg_objName.bin
         Inode      node at 0x001c832c, totlen 0x00000303, #ino    448, version     1, isize     4883, csize      703, dsize     4096, offset        0
         Inode      node at 0x001c8630, totlen 0x000000fd, #ino    448, version     2, isize     4883, csize      185, dsize      787, offset     4096
```

I am not familiar with JFFS2 internals but https://www.sourceware.org/jffs2/jffs2-html/node3.html#SECTION00033000000000000000 states that:
> In order to facilitate rapid decompression of data upon readpage() requests, nodes contain no more than a single page of data, according to the hardware page size on the target platform.

After having a look at `jffs2reader.c` I'm quite sure that it's not able to extract files split among several nodes. Maybe `jffs2reader` should go and check if there are other inodes with the same inode # right after the inode and extract them?

Note: `zcfg_objName.bin` is also split among two inodes. Only the fist one is extracted. However jffs2dump does not report CRC errors or empty space.
Note²: When running `jffs2reader 10000.jffs2 -d / -r` jffs2reader is stuck in an endless loop, 100% CPU usage.

devttys0, 2015-02-18:
I suspect the NULL bytes after `0x1000` are due to the default JFFS2 page size, which limits the maximum node size; from the `mkfs.jffs2` help output:

```
-s, --pagesize=SIZE     Use page size (max data node size) SIZE.
                        Set according to target system's memory management
                        page size (default: 4KiB)
```

You should be able to specify the page size manually when mounting a JFFS2 image, but I don't think `jffs2reader` has such an option.

The infinite loop in `jffs2reader` should definitely be reported upstream to the `mtd-utils` project, possibly with a request/patch to specify page sizes as well, if that is in fact the issue here.

sviehb, 2015-03-10:
I am very confident that the `jffs2reader` implementation is incomplete when it comes to compression/file contents that are split among several nodes. I have jet to see a JFFS2 extraction that works for compressed files >0x1000!
I threw together a Python-only JFFS2 extraction tool which can be found here: https://github.com/sviehb/jefferson
`jefferson` handles big/litte endian filesystems, compression, concatenated filesystems,CRC and is reasonably fast. I'm sure there are still a few bugs, but I think it would be a good fit for `binwalk`.

sviehb, 2015-03-13:
I did some tests using:
`^jffs2 filesystem:jffs2:/path_to_jefferson/jefferson.py -d 'jffs2-root' '%e'`
instead of the `unjffs2.py` plugin. Found a few bugs which should now be fixed.
I used the hardcoded destination path `jffs2-root` because I want to keep `binwalk` from trying to extracting each occurrence of JFFS2 individually. `jefferson` now immediately exits if the destination directory already exists.

However it's not very elegant to integrate it like that. `binwalk` still writes a file and calls `jefferson` for every single JFFS2 match (which can be 100s). It would be nice to have a flag in `extract.conf` that indicates that the extractor should only be used once per file.

Here is a firmware sample that contains multiple JFFS2 filesystems:
http://www.downloads.netgear.com/files/GDC/DGND4000/DGND4000_V1.1.00.14_1.00.14NA.zip

devttys0, 2015-03-25:
The multiple JFFS2 entries should never have happened in the first place; the signatures didn't take into account padding between JFFS2 nodes, and would erroneously mark some valid JFFS2 nodes as invalid. This resulted in binwalk thinking there were multiple JFFS2 nodes with some 'invalid' data in between.

I've added a plugin to handle this, so the multiple JFFS2 results should now be a thing of the past:

```
$ binwalk DGND4000.bin 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
16314         0x3FBA          Sercomm firmware signature, version control: 0, download control: 256, hardware ID: "AGP", hardware version: 0x4100, firmware version: 0x14, starting code segment: 0x0, code size: 0x7300
16377         0x3FF9          Sercomm firmware signature, version control: 0, download control: 0, hardware ID: "", hardware version: 0x0, firmware version: 0x0, starting code segment: 0x0, code size: 0x0
131072        0x20000         JFFS2 filesystem, big endian
```

What would be the best way to integrate jefferson into binwalk? I've added `jefferson.py` as an extraction rule, which works fine when you just copy `jefferson.py` into a $PATH directory. But with the new jefferson lzma python module, that seems a bit messy. It could be implemented as a binwalk plugin, but then there's duplication of code. I think if jefferson included a `setup.py` for 'proper' installation, that would be ideal.

sviehb, 2015-03-27:
Just added a `setup.py` to `jefferson` and dropped the `.py` for the script.
`^jffs2 filesystem:jffs2:jefferson -d '%%jffs2-root%%' '%e':0:False`

Btw you could check the node header (`hdr_crc`) value in  `jffs2valid`. That's the CRC of just 8 bytes (CRC flavor in `mtd_crc()` in `jefferson`) so should be faster than looping around in 10k of memory in Python.

devttys0, 2015-03-27:
Done and done, thanks!

closed 2015-03-27 by devttys0