ReFirmLabs/binwalk #110: file command broken, after installing Binwalk

kamil-HFC, 2015-02-19:
Hi. Yesterday after installing Binwalk, i can't use file command.
Removig both of them- file and Binwalk did't help. 
Earilier everything was fine.

$ file setup.py 
file: symbol lookup error: file: undefined symbol: magic_version

$ strace file setup.py 
execve("/usr/bin/file", ["file", "setup.py"], [/* 68 vars */]) = 0
brk(0)                                  = 0x1f5a000
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f833dc4b000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=134959, ...}) = 0
mmap(NULL, 134959, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f833dc2a000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libmagic.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\240B\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=464741, ...}) = 0
mmap(NULL, 2209648, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f833d80f000
mprotect(0x7f833d829000, 2097152, PROT_NONE) = 0
mmap(0x7f833da29000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1a000) = 0x7f833da29000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\320\37\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1845024, ...}) = 0
mmap(NULL, 3953344, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f833d449000
mprotect(0x7f833d604000, 2097152, PROT_NONE) = 0
mmap(0x7f833d804000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1bb000) = 0x7f833d804000
mmap(0x7f833d80a000, 17088, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f833d80a000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libz.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\"\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=113383, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f833dc29000
mmap(NULL, 2200104, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f833d22f000
mprotect(0x7f833d247000, 2097152, PROT_NONE) = 0
mmap(0x7f833d447000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x18000) = 0x7f833d447000
close(3)                                = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f833dc28000
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f833dc26000
arch_prctl(ARCH_SET_FS, 0x7f833dc26740) = 0
mprotect(0x7f833d804000, 16384, PROT_READ) = 0
mprotect(0x7f833d447000, 4096, PROT_READ) = 0
mprotect(0x7f833da29000, 4096, PROT_READ) = 0
mprotect(0x603000, 4096, PROT_READ)     = 0
mprotect(0x7f833dc4d000, 4096, PROT_READ) = 0
munmap(0x7f833dc2a000, 134959)          = 0
brk(0)                                  = 0x1f5a000
brk(0x1f7b000)                          = 0x1f7b000
open("/usr/lib/locale/locale-archive", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=4167216, ...}) = 0
mmap(NULL, 4167216, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f833ce35000
close(3)                                = 0
writev(2, [{"file", 4}, {": ", 2}, {"symbol lookup error", 19}, {": ", 2}, {"file", 4}, {": ", 2}, {"undefined symbol: magic_version", 31}, {"", 0}, {"", 0}, {"\n", 1}], 10file: symbol lookup error: file: undefined symbol: magic_version
) = 65
exit_group(127)                         = ?
+++ exited with 127 +++

devttys0, 2015-02-19:
What version of `binwalk` did you install? The latest code in the git repository does not require `libmagic` at all.

Did you install `libmagic` manually? This issue is usually a problem with a mis-match between your `file` version and the `libmagic` version. Since both `file` and `libmagic` are contained in the same source download, installing both of them from the same source download should fix this issue.

kamil-HFC, 2015-02-20:
I'm using latest ver of Binwalk. Your tip solved my problem. Thank You.

closed 2015-02-20 by kamil-HFC