ReFirmLabs/binwalk #114: install Capstone 3.0.1 as dependency

aquynh, 2015-02-23:
from the latest version **3.0.1**, it is possible to install Capstone engine just from **PyPi**, like:

    pip install capstone

unlike previous versions, this also downloads, compiles & builds the core (`libcapstone.so`), so it is no longer necessary to build & install the core separately like before. 

this means if you only use Capstone from **Python**, above command is enough to *install everything* you need to use Capstone.

devttys0, 2015-02-25:
Installing capstone for Python3 using pip seems to be broken (haven't tried Python2):

```
$ sudo pip3 install capstone
$ binwalk 
Traceback (most recent call last):
  File "/usr/local/bin/binwalk", line 26, in <module>
    import binwalk.modules
  File "/usr/local/lib/python3.4/dist-packages/binwalk/modules/__init__.py", line 3, in <module>
    from binwalk.modules.disasm import Disasm
  File "/usr/local/lib/python3.4/dist-packages/binwalk/modules/disasm.py", line 1, in <module>
    import capstone
  File "/usr/local/lib/python3.4/dist-packages/capstone/__init__.py", line 286, in <module>
    ctypes.c_uint64, ctypes.c_size_t, ctypes.POINTER(ctypes.POINTER(_cs_insn)))
  File "/usr/local/lib/python3.4/dist-packages/capstone/__init__.py", line 281, in _setup_prototype
    getattr(lib, fname).restype = restype
  File "/usr/lib/python3.4/ctypes/__init__.py", line 364, in __getattr__
    func = self.__getitem__(name)
  File "/usr/lib/python3.4/ctypes/__init__.py", line 369, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: /usr/lib/libcapstone.so: undefined symbol: cs_disasm
```

Everything works fine after building capstone from source and installing the Python3 bindings. Could be user error on my part?

aquynh, 2015-02-25:
1) can you copy `bindings\python\test.py` to an isolated place, and run it with Python3 then report back the result?

2) can you try to install with `pip`, then test `binwalk` again with Python2? Python2 is better tested, so i suppose it has no problem.

aquynh, 2015-02-27:
i can confirm that (1) has no issue on my Ubuntu machine: `pip3` & `python3` works perfectly here.

i havent tried with `binwalk`, but perhaps something is wrong on your side?

devttys0, 2015-03-02:
```
perhaps something is wrong on your side
```

Must have been; I re-installed capstone for both Python2 and Python3 via pip and now it all works fine. Documentation updated accordingly!

closed 2015-03-02 by devttys0