ReFirmLabs/binwalk #115: virtualenv + binwalk (2.1.0)

7histle, 2015-03-01:
```
tools@wrklpt:~$ source ~/tools_env/bin/activate
(tools_env)tools@wrklpt:~$ pip install binwalk
Collecting binwalk
  Using cached binwalk-2.1.0.tar.gz
Installing collected packages: binwalk
  Running setup.py install for binwalk
    changing mode of build/scripts-2.7/binwalk from 664 to 775
    changing mode of /home/tools/tools_env/bin/binwalk to 775
Successfully installed binwalk-2.1.0
(tools_env)tools@wrklpt:~$ binwalk /tmp/yandex_shit/chua_list
Traceback (most recent call last):
  File "/home/tools/tools_env/bin/binwalk", line 25, in <module>
    import binwalk
  File "/home/tools/tools_env/local/lib/python2.7/site-packages/binwalk/__init__.py", line 3, in <module>
    from binwalk.core.module import Modules, ModuleException
ImportError: No module named core.module
(tools_env)tools@wrklpt:~$
```
**ImportError: No module named core.module**

Info
```
(tools_env)tools@wrklpt:~$ cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=14.04
DISTRIB_CODENAME=trusty
DISTRIB_DESCRIPTION="Ubuntu 14.04.2 LTS"
(tools_env)tools@wrklpt:~$
```

devttys0, 2015-03-02:
I don't maintain binwalk in pypy, and last I heard it was broken. 

Binwalk does work when installed to a prefix directory using `setup.py`, so I'm not sure if this is an issue with binwalk, pip, or some environment-specific issue; however, it seems odd that the `binwalk` module was found, but not `binwalk.core.module`. Have you verified that `/home/tools/tools_env/local/lib/python2.7/site-packages/binwalk/core/module.py` exists?

7histle, 2015-03-08:
Sorry for long reply.
```
 pip install binwalk
Collecting binwalk
  Using cached binwalk-2.1.0.tar.gz
Installing collected packages: binwalk
  Running setup.py install for binwalk
    changing mode of build/scripts-2.7/binwalk from 664 to 775
    changing mode of /home/tools/tools_env/bin/binwalk to 775
Successfully installed binwalk-2.1.0
(tools_env)tools@wrklpt:~$ cd /home/tools/tools_env/local/lib/python2.7/site-packages/binwalk
(tools_env)tools@wrklpt:~/tools_env/local/lib/python2.7/site-packages/binwalk$ ls -alh
total 16K
drwxrwxr-x  2 tools tools 4.0K Mar  8 10:51 .
drwxrwxr-x 23 tools tools 4.0K Mar  8 10:51 ..
-rw-rw-r--  1 tools tools  292 Mar  8 10:51 __init__.py
-rw-rw-r--  1 tools tools  713 Mar  8 10:51 __init__.pyc
(tools_env)tools@wrklpt:~/tools_env/local/lib/python2.7/site-packages/binwalk$ #:|
```

devttys0, 2015-03-09:
Yeah, it definitely looks like the pip install is borked. You should be able to grab the latest code from the master branch and build/install it in your environment just with the standard commands:

```
$ python setup.py install
```

closed 2015-03-25 by devttys0

kaleemullah360, 2019-08-26:
While working on **Ubuntu** and created **virtual environment** for my **Django project**.

**setup binwalk without virtual environment via following tutorial:**

[https://github.com/ReFirmLabs/binwalk/blob/master/INSTALL.md](https://github.com/ReFirmLabs/binwalk/blob/master/INSTALL.md)

**first Install dependencies required for binwalk to work**

Download & Install latest **Binwalk** from **git** :
>_

```
cd ~
wget https://github.com/devttys0/binwalk/archive/master.zip
unzip master.zip
cd ~/binwalk-master && sudo python3 setup.py uninstall && sudo python3 setup.py install
```

To check where binwalk is installed do
>_
`which binwalk`
it will show following directory, where binwalk is installed:

> /usr/local/bin/binwalk

**Setup Binwalk for project Virtual Environment:**
while virtual environment activated, install binwalk via pip.
`pip3 install binwalk`

try importing binwalk in python shell:
>_
```
python3
import binwalk
```

**it raises binwalk core module missing error.**
 to resolve it remove existing binwalk directory from virtual environment python packages.
to locate binwalk directory, first activate virtual environment and do
>_
`which binwalk`
it will show following directory, where binwalk is installed in virtual environment:

> /home/user/.virtualenves/my_project/lib/python3.7/site-packeges/binwalk

remove existing binwalk directory

`rm -R /home/user/.virtualenves/my_project/lib/python3.7/site-packeges/binwalk`

copy binwalk directory to virtual environment packages directory.  

`cp /usr/local/bin/binwalk /home/user/.virtualenves/my_project/lib/python3.7/site-packeges/`