ReFirmLabs/binwalk #116: Trouble extracting JFFS2

Paul1989, 2015-03-12:
Hi,

First off, I'm quite new to firmware reverse engineering so please accept my apologies if this issue is user error as opposed to a problem with Binwalk.
I'm trying to extract the contents of this http://is.gd/ew0u8E file using the latest version of Binwalk, Python3 & Wheezy. Running binwalk without -I results in nothing being found & including -I results in the below:

DECIMAL       HEXADECIMAL     DESCRIPTION                                       
--------------------------------------------------------------------------------
3466          0xD8A           JFFS2 filesystem, little endian                   
4908          0x132C          BFF volume entry, AIXv3, file name: "ê?NÈEÈïÌHô3þÁ
Q.Qÿ¡nßCàÎnJ§M´ÖeFÜp%ÃJã2•X«"s²É!Ó8ÐëÅøÖ£[¾k«kpIôÿÝù²¬ÓÛ9&LÀ nÈ*§<¢Ê
³ò³Á¾½w tFÖùx"                                                                
5703          0x1647          JFFS2 filesystem, little endian                   

Adding -e results in:

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
Wrong bitmask  at  0x00000000, 0x8519
Wrong bitmask  at  0x00000004, 0xec03
Wrong bitmask  at  0x00000008, 0xd318
Wrong bitmask  at  0x0000000c, 0xa00e
(Trimmed as there are around 375000 lines like the above)

The firmware is stored in flash so JFFS2 seems sensible, but as you may have guessed from the error, the generated JFFS2 folder is empty.
Any hints about the above errors would be very much appreciated - as per above I'm sorry if it turns out to be an issue with something other than Binwalk.

Cheers.

sviehb, 2015-03-13:
Your firmware sample looks like it's encrypted. The results you saw with the `--invalid` switch are just false positives.
The encryption method used is most likely AES ECB. Why? When looking at the file in a hex editor you will frequently see 16-byte-blocks with the contents `14 4F 90 52 78 15 D7 7C 40 C7 43 24 3E F8 E3 E5` or `0D 49 11 0D EE 13 BA 0D A1 00 AF E5 42 55 CC A8`. These are most likely encrypted 0x00 or 0xFF blocks and indicate ECB mode.

Paul1989, 2015-03-13:
Hi,

Many thanks for taking the time to research this for me. Unfortunate but interesting never the less.

Cheers,
Paul.

closed 2015-03-25 by devttys0