ReFirmLabs/binwalk #121: ImportError: No module named core.module

dejakke, 2015-04-04:
Hi,

I'm trying to install binwalk on a slackware-machine.
And although i can see that during the installation the file ../core/modules.py is actually installed.
Upon running it's seems that my system is not able to find this file!
Could this be just a simple python-path issue, or is there something else wrong?
For debugging purposes i'll add the log here.

bash-4.2$ binwalk
Traceback (most recent call last):
  File "/usr/local/bin/binwalk", line 17, in <module>
    import binwalk
  File "build/bdist.linux-x86_64/egg/binwalk/__init__.py", line 3, in <module>
    # $Id$
ImportError: No module named core.module
bash-4.2$

devttys0, 2015-04-04:
The error message looks a bit odd / incomplete (e.g., there is no `init.py`; it should be `__init__.py`); it looks like it might have been installed via a Python egg package? How was binwalk installed?

dejakke, 2015-04-05:
Well, this was done by using the git-repo first with the latest commit and afterwards to the commit tag v2.0.1.
And it seems the github has some issues with the '__' and '<' '>' characters for displaying comments.
but the error message indeeds complains about the file '__init__.py, line 3, in <module>

devttys0, 2015-04-05:
Ah, sorry; you need to wrap text around three backticks (```) to get it displayed as "raw" text.

Still, I'm surprised that your install path is `build/bdist.linux-x86_64/egg/binwalk/`; usually the python installer will put things in `python2.7/dist-packages/binwalk/`, or something similar (maybe this is a slackware thing?).

In any case, I'd double-check that `setup.py` is actually installing files to the `build/bdist.linux-x86_64/egg/binwalk/` directory, and that `build/bdist.linux-x86_64/egg/binwalk/core/module.py` exists.

devttys0, 2015-05-04:
Were you able to resolve this issue, or is it still something that needs attention?

dejakke, 2015-05-14:
I've just finished a reinstall of the os, and it seems to work now.
Guess something had screwed up my python path or something.
So yes, this issue can be closed. :)

closed 2015-05-14 by dejakke

ytrezq, 2016-09-03:
@devttys0 : no please reopen !

I found how to always reproduce it :

    pip install binwalk
will always trigger this on cygwin.

devttys0, 2016-09-04:
The binwalk package on [PyPi](https://pypi.python.org/pypi/binwalk) is not managed by myself, is outdated, not maintained, and known to be broken. Remove it and [install from source](https://github.com/devttys0/binwalk/wiki/Quick-Start-Guide) and this error should disappear.

EvanCarroll, 2019-01-24:
Can we change this issue then to be properly named?

> Binwalk on PyPi is unsupported.

Perhaps we can remove binwalk from PyPi if you don't want to use it for distribution (though it seems better to just keep the package up to date).