ReFirmLabs/binwalk #122: Some PDF file does not identify by binwalk

fact0real, 2015-04-07:
Hello, 
In this exe [file], there is one PDF file that binwalk does not identify it, please check it.
thanks
[file]:http://a.asis.io/mdWTHrzu.exe

devttys0, 2015-04-07:
I see only one location (offset `0x6F`) that appears to have a PDF header, and binwalk identifies it properly:

```
$ binwalk mdWTHrzu.exe 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             Microsoft executable, portable (PE)
111           0x6F            PDF document, version: "1.5"
463016        0x710A8         Unix path: /sT/sT/sT/sT0sT1tU1tU1tU2uV2uV2uV2uV2uV2uV2uV2uV2uV2uV2uV2uV2vW3wX3wX3wX4xY4xY4xY4xY3wX3wX3wX3wX3wX3wX3wX3wX2xY0yY0yY0yY0yY0yY0y
466013        0x71C5D         Unix path: /rS/rS/rS/rS/rS/rS/rS/rS0sT0sT0sT1tU1tU1tU1tU1tU1tU1tU1tU1tU1tU2uV1uV1uV2vW2vW2vW3wX3wX2vW2vW2vW2vW2vW2vW2vW2vW1wX1wX1wX1wX1wX1w
496128        0x79200         Zip archive data, at least v1.0 to extract, name: Stegano-BMP-master/
496186        0x7923A         Zip archive data, at least v1.0 to extract, compressed size: 187, uncompressed size: 564, name: Stegano-BMP-master/Makefile
496439        0x79337         Zip archive data, at least v1.0 to extract, compressed size: 144, uncompressed size: 238, name: Stegano-BMP-master/README.md
496650        0x7940A         Zip archive data, at least v1.0 to extract, compressed size: 310, uncompressed size: 533, name: Stegano-BMP-master/decrypt.cpp
497029        0x79585         Zip archive data, at least v1.0 to extract, name: Stegano-BMP-master/easybmp/
497095        0x795C7         Zip archive data, at least v1.0 to extract, compressed size: 782, uncompressed size: 1489, name: Stegano-BMP-master/easybmp/BSD_(revised)_license.txt
497968        0x79930         Zip archive data, at least v1.0 to extract, compressed size: 9338, uncompressed size: 48149, name: Stegano-BMP-master/easybmp/EasyBMP.cpp
507383        0x7BDF7         Zip archive data, at least v1.0 to extract, compressed size: 840, uncompressed size: 2418, name: Stegano-BMP-master/easybmp/EasyBMP.h
508298        0x7C18A         Zip archive data, at least v1.0 to extract, compressed size: 823, uncompressed size: 2815, name: Stegano-BMP-master/easybmp/EasyBMP_BMP.h
509200        0x7C510         Zip archive data, at least v1.0 to extract, compressed size: 9396, uncompressed size: 31370, name: Stegano-BMP-master/easybmp/EasyBMP_ChangeLog.txt
518683        0x7EA1B         Zip archive data, at least v1.0 to extract, compressed size: 973, uncompressed size: 2708, name: Stegano-BMP-master/easybmp/EasyBMP_DataStructures.h
519746        0x7EE42         Zip archive data, at least v1.0 to extract, compressed size: 558, uncompressed size: 1975, name: Stegano-BMP-master/easybmp/EasyBMP_VariousBMPutilities.h
520399        0x7F0CF         Zip archive data, at least v1.0 to extract, name: Stegano-BMP-master/easybmp/sample/
520472        0x7F118         Zip archive data, at least v1.0 to extract, compressed size: 180430, uncompressed size: 308278, name: Stegano-BMP-master/easybmp/sample/EasyBMPbackground.bmp
700996        0xAB244         Zip archive data, at least v1.0 to extract, compressed size: 820, uncompressed size: 2835, name: Stegano-BMP-master/easybmp/sample/EasyBMPsample.cpp
701906        0xAB5D2         Zip archive data, at least v1.0 to extract, compressed size: 5599, uncompressed size: 113266, name: Stegano-BMP-master/easybmp/sample/EasyBMPtext.bmp
707593        0xACC09         Zip archive data, at least v1.0 to extract, compressed size: 707, uncompressed size: 1807, name: Stegano-BMP-master/easybmp/sample/makefile
708381        0xACF1D         Zip archive data, at least v1.0 to extract, compressed size: 368, uncompressed size: 768, name: Stegano-BMP-master/encrypt.cpp
708818        0xAD0D2         Zip archive data, at least v1.0 to extract, compressed size: 28, uncompressed size: 28, name: Stegano-BMP-master/main.cpp
708912        0xAD130         Zip archive data, at least v1.0 to extract, compressed size: 937, uncompressed size: 2705, name: Stegano-BMP-master/stegano.cpp
709918        0xAD51E         Zip archive data, at least v1.0 to extract, compressed size: 137, uncompressed size: 196, name: Stegano-BMP-master/stegano.h
712075        0xADD8B         End of Zip archive, comment: "6e73434c01138685e9ade26e76724b13f9d9b663"
712607        0xADF9F         Unix path: /Type/XObject/Subtype/Image/Width 550 /Height 367 /BitsPerComponent 8 /ColorSpace/DeviceRGB/Filter/DCTDecode/Length 48480>>
761311        0xB9DDF         Unix path: /S/Transparency/CS/DeviceRGB/K true>>
772045        0xBC7CD         Unix path: /Type/FontDescriptor/FontName/BAAAAA+LiberationSerif
772655        0xBCA2F         Unix path: /Type/Font/Subtype/TrueType/BaseFont/BAAAAA+LiberationSerif
773040        0xBCBB0         Unix path: /PDF/Text/ImageC/ImageI/ImageB]
773163        0xBCC2B         Unix path: /S/Transparency/CS/DeviceRGB/I true>>/Contents 2 0 R>>
```

Is there another PDF that I'm missing?

fact0real, 2015-04-07:
But I get this:
```
$ binwalk mdWTHrzu.exe

WARNING: Failed to load plugin module 'compressd': Failed to locate libraries '['compress42']'

WARNING: Failed to load plugin module 'zlibvalid': Failed to locate libraries '['tinfl']'

WARNING: Failed to load plugin module 'compressd': Failed to locate libraries '['compress42']'

WARNING: Failed to load plugin module 'zlibvalid': Failed to locate libraries '['tinfl']'

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             Microsoft portable executable
9764          0x2624          Zlib compressed data, default compression
12168         0x2F88          Zlib compressed data, compressed
80213         0x13955         Zlib compressed data, default compression
154469        0x25B65         Zlib compressed data, default compression
179066        0x2BB7A         Zlib compressed data, default compression
184700        0x2D17C         Zlib compressed data, default compression
186509        0x2D88D         Zlib compressed data, default compression
198460        0x3073C         Zlib compressed data, default compression
202685        0x317BD         Zlib compressed data, default compression
231188        0x38714         Zlib compressed data, default compression
358799        0x5798F         Zlib compressed data, default compression
373790        0x5B41E         Zlib compressed data, default compression
397556        0x610F4         Zlib compressed data, default compression
496128        0x79200         Zip archive data, at least v1.0 to extract, name: "Stegano-BMP-master/"
496186        0x7923A         Zip archive data, at least v1.0 to extract, compressed size: 187,  uncompressed size: 564, name: "Stegano-BMP-master/Makefile"
496439        0x79337         Zip archive data, at least v1.0 to extract, compressed size: 144,  uncompressed size: 238, name: "Stegano-BMP-master/README.md"
496650        0x7940A         Zip archive data, at least v1.0 to extract, compressed size: 310,  uncompressed size: 533, name: "Stegano-BMP-master/decrypt.cpp"
497029        0x79585         Zip archive data, at least v1.0 to extract, name: "Stegano-BMP-master/easybmp/"
497095        0x795C7         Zip archive data, at least v1.0 to extract, compressed size: 782,  uncompressed size: 1489, name: "Stegano-BMP-master/easybmp/BSD_(revised)_license.txt"
497968        0x79930         Zip archive data, at least v1.0 to extract, compressed size: 9338,  uncompressed size: 48149, name: "Stegano-BMP-master/easybmp/EasyBMP.cpp"
507383        0x7BDF7         Zip archive data, at least v1.0 to extract, compressed size: 840,  uncompressed size: 2418, name: "Stegano-BMP-master/easybmp/EasyBMP.h"
508298        0x7C18A         Zip archive data, at least v1.0 to extract, compressed size: 823,  uncompressed size: 2815, name: "Stegano-BMP-master/easybmp/EasyBMP_BMP.h"
509200        0x7C510         Zip archive data, at least v1.0 to extract, compressed size: 9396,  uncompressed size: 31370, name: "Stegano-BMP-master/easybmp/EasyBMP_ChangeLog.txt"
518683        0x7EA1B         Zip archive data, at least v1.0 to extract, compressed size: 973,  uncompressed size: 2708, name: "Stegano-BMP-master/easybmp/EasyBMP_DataStructures.h"
519746        0x7EE42         Zip archive data, at least v1.0 to extract, compressed size: 558,  uncompressed size: 1975, name: "Stegano-BMP-master/easybmp/EasyBMP_VariousBMPutilities.h"
520399        0x7F0CF         Zip archive data, at least v1.0 to extract, name: "Stegano-BMP-master/easybmp/sample/"
520472        0x7F118         Zip archive data, at least v1.0 to extract, compressed size: 180430,  uncompressed size: 308278, name: "Stegano-BMP-master/easybmp/sample/EasyBMPbackground.bmp"
700996        0xAB244         Zip archive data, at least v1.0 to extract, compressed size: 820,  uncompressed size: 2835, name: "Stegano-BMP-master/easybmp/sample/EasyBMPsample.cpp"
701906        0xAB5D2         Zip archive data, at least v1.0 to extract, compressed size: 5599,  uncompressed size: 113266, name: "Stegano-BMP-master/easybmp/sample/EasyBMPtext.bmp"
707593        0xACC09         Zip archive data, at least v1.0 to extract, compressed size: 707,  uncompressed size: 1807, name: "Stegano-BMP-master/easybmp/sample/makefile"
708381        0xACF1D         Zip archive data, at least v1.0 to extract, compressed size: 368,  uncompressed size: 768, name: "Stegano-BMP-master/encrypt.cpp"
708818        0xAD0D2         Zip archive data, at least v1.0 to extract, compressed size: 28,  uncompressed size: 28, name: "Stegano-BMP-master/main.cpp"
708912        0xAD130         Zip archive data, at least v1.0 to extract, compressed size: 937,  uncompressed size: 2705, name: "Stegano-BMP-master/stegano.cpp"
709918        0xAD51E         Zip archive data, at least v1.0 to extract, compressed size: 137,  uncompressed size: 196, name: "Stegano-BMP-master/stegano.h"
712075        0xADD8B         End of Zip archive, comment: "6e73434c01138685e9ade26e76724b13f9d9b663"
742198        0xB5336         Zlib compressed data, default compression
758682        0xB939A         Zlib compressed data, default compression
763507        0xBA673         Zlib compressed data, best compression
776073        0xBD789         Zlib compressed data, best compression`
```

fact0real, 2015-04-07:
When I upgrade from 2.0.0 to 2.1.0, it seems that this problem was solved.

closed 2015-04-08 by devttys0