ReFirmLabs/binwalk #125: "ubelong" keyword fails when used in magics at offset 0 (but "belong" works)

JElchison, 2015-05-04:
Using latest stable release (v2.0.1, 57789cd)

Environment:
```
$ uname -a
Linux xxxxx 3.13.0-51-generic #84-Ubuntu SMP Wed Apr 15 12:11:46 UTC 2015 i686 i686 i686 GNU/Linux
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.2 LTS
Release:	14.04
Codename:	trusty
```

Also observed on Ubuntu 12.04.5 LTS.

Setup:
```
echo -n 0x7FFF000022222222 | xxd -r -p > jjj
echo "0  belong  0x7FFF0000  JJJ" > magic.txt
echo "0  ubelong  0x7FFF0000  JJJ" > umagic.txt
```

Sanity check:
```
$ xxd -g4 jjj
0000000: 7fff0000 22222222                    ....""""
```

Test 1 (expected behavior):
```
$ binwalk -m magic.txt jjj

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             JJJ
```

Test 2 (unexpected behavior):
```
$ binwalk -m umagic.txt jjj

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
```

I don't understand why Test 2 doesn't succeed in identifying the file as "JJJ".  Am I doing something wrong?

This misbehavior still shows up when using values other than 0x7FFF0000.

Thanks!

devttys0, 2015-05-04:
It doesn't look like you're doing anything wrong; this is probably a bug in v2.0.1. Both `belong` and `ubelong` should work as expected (at least, they do for me) using the latest code from the master branch.

JElchison, 2015-05-04:
Confirmed, version 02dd825 does not manifest this problem.  Feel free to close this issue, at your discretion.

closed 2015-05-04 by devttys0