ReFirmLabs/binwalk #126: unrecognized file type

gcherogiorgos, 2015-05-12:
Hi,

I downloaded the latest firmware for the Buffalo WBMR-HP-G300H wireless modem/router, unzipped it and ran binwalk against it, but binwalk came up empty. I also ran strings to see if I'd get any useful output, but still nothing.

The modem's page is http://goo.gl/tGtF1E

And you can download the firmware from http://goo.gl/42DdmS

Thank you very much for your time.

George

PS: The unzipped firmware is 24.6 MB. Isn't it a bit strange?

devttys0, 2015-05-13:
Aside from some obvious header data at the very beginning of the file, the entropy of the firmware image suggests that it is encrypted (run `binwalk -E` to see the entropy). This hypothesis is supported by other sources, who claim that Buffalo encrypts their firmware updates (http://www.dd-wrt.com/wiki/index.php/WHR-G300N).

The size is abnormally large for an ADSL modem, but not outside the realm of possibility. I've seen other routers with 64/128MB of flash, or even more in some cases.

gcherogiorgos, 2015-05-13:
Thank you very much for your prompt response. I appreciate it.

Regards,

George

closed 2015-05-13 by gcherogiorgos