ReFirmLabs/binwalk #127: Extracts into target file directory, not current directory

rogerbinns, 2015-05-15:
I put a target file in /tmp, and then in a subdirectory ran binwalk -e /tmp/targetfw

The exit status was 0, but there was nothing extracted.  Eventually I managed to work out that the extraction was alongside the original target file, and not the current directory.  (Note that tar, zip, cpio, 7z, etc all extract into the current directory hence the expectation.)

At the very least I suggest updating the documentation (man page and wiki page) to mention this.  It may also be worthwhile having a message that says how many files were extracted and where, although that is a little contrary to the unix philosophy of being silent on success.

devttys0, 2015-05-18:
Agreed; pushed patch to output all files to CWD.

ddcc, 2015-06-03:
Would it be possible to allow the user to modify the output directory, e.g. using an additional argument? I have a multithreaded application using the Binwalk API that concurrently extracts multiple firmware images each to a unique temporary directory, but this now fails since the current working directory is per-process and not per-thread.

devttys0, 2015-07-16:
There is now a `--directory` option that lets you specify a directory other than CWD.

closed 2015-07-16 by devttys0