ReFirmLabs/binwalk #129: UBIFS Packed Images

matthew1471, 2015-05-24:
I noticed while looking at one of my firmwares that the image was an UBI FS file system (http://www.dubeiko.com/development/FileSystems/UBI/ubidesign.pdf, file magic on page 9, "UBI#" and "UBI!"). 

There are no user space tools at present (http://www.linux-mtd.infradead.org/faq/ubifs.html#L_ubifs_extract) but using a virtual memory device one should be able to extract the data.

I'm creating this issue for all things binwalk and UBI related :-). Firstly I think it would help if binwalk could be updated to spot UBIFS file magic?

For those also looking, there's a "nandsim" and "mtdram" that simulate these devices, then one should in theory be able to use ubiattach to get closer to mounting.. this pae was helpful : http://pjankows.blogspot.co.nz/2012/01/how-to-mount-ubi-image.html

However after trimming to the first file magic and researching how much flash the device has (32MB) I cannot get the right options to mount it, does anyone have any experience that could help me :-)?

Thanks,
Matthew

devttys0, 2015-05-24:
The current master branch has UBIFS [signatures](https://github.com/devttys0/binwalk/blob/master/src/binwalk/magic/filesystems#L108); if these are not correctly identifying the UBI file system(s), that obviously needs correction.

I've not dealt directly with UBIFS myself, but [infradead.org](http://www.linux-mtd.infradead.org/faq/ubi.html) suggests that your particular error is due to an incorrect sub-page size:

```
To fix this, you should override the default sub-page size that ubiformat uses to what the kernel expects using the -s option of ubiformat.
```

Perhaps `nandsim`, `mtdram`, or `ubiattach` have some similar option?

matthew1471, 2015-05-25:
Ah thanks, I forgot how out of date Kali Linux was.

Thank you also for your kind help trying to get the image mounted. I tried it but it still did not work:

```
# Try mtdram
modprobe mtdram total_size=32000
cat /proc/mtd
flash_erase /dev/mtd0 0 0
ubiformat /dev/mtd0 -f UBI.bin
modprobe ubi
ubiattach -p /dev/mtd0
dmesg
```

>...bad VID header offset 512, expected 64..

```
flash_erase /dev/mtd0 0 0
ubiformat /dev/mtd0 -f UBI.bin -s 64
```

>ubiformat: error! : sub-page cannot be larger than min. I/O unit

```
modprobe -r mtdram
```

```
# Try nandsim
modprobe nandsim first_id_byte=0x20 second_id_byte=0x35
cat /proc/mtd
flash_erase /dev/mtd0 0 0
ubiformat /dev/mtd0 -f UBI.bin -s 64
```
>...min. I/O size 512 bytes...
>ubiformat: error!: bad UBI magic 00000000, should be 0x55424923
>ubiformat: error!: bad EC header at eraseblock 1 of "UBI.bin"

```
flash_erase /dev/mtd0 0 0
ubiformat /dev/mtd0 -f UBI.bin
flash_erase /dev/mtd0 0 0
```

I tried all manner of s values, they couldn't be bigger than the min. I/O size (which was 1 when using mtdram).. previously when I tried nandsim I also looked at all the different NANDs it supported and tried to find one with a smaller block size but couldn't.

I believe nandsim is for NAND memory and mtdram is for NOR memory.. I seem to get closer with mtdram (it's reading the file magic and the header correctly!). The original device does not list what model flash it uses (other than that it's 32MB) and it's too expensive and in warranty to take apart.

Any other suggestions are very welcome (I've been Googling for ages and haven't read anyone having much success with this).

sviehb, 2015-09-21:
#154 adds UBIFS extraction capabilities. `UBIValidPlugin` validates UBI headers and dynamically calculates jump offsets. Only the fist valid occurrence of an `UBI erase count header` per file is displayed. All other valid occurrences are jumped over (`result.jump`). 
In order to initially determine the eraseblock size (PEB size) the plugin has to preserve some state. PEB size is usually 128k. I'm not sure how common other sizes are.

The `ubi_reader` project is used for extraction. The UBI/UBIFS format is quite complex and `ubi_reader` still has some issues and is not very pretty in general.. but it's a fist step towards reliable, OS-independent extraction.

closed 2018-04-12 by CoffeeExpress