ReFirmLabs/binwalk #130: Improve shared/dynamic library search on OS X

UniqMartin, 2015-06-01:
One of the easiest and probably most common way to obtain `binwalk` on OS X is via [Homebrew](http://brew.sh/). However, the experience is suboptimal for people with a non-standard installation (i.e. install location is not `/usr/local`). I'd like to help address that.

The problem is inside [src/binwalk/core/C.py, lines 128–131](https://github.com/devttys0/binwalk/blob/c5bbfd631f1d1055ec418087457f2e488f22aafd/src/binwalk/core/C.py#L128-131), that searches the following directories on OS X:

- binwalk's private `libs` directory (no longer needed since there are no longer any bundled libraries?).
- `/opt/local/lib` (standard MacPorts installation).
- `/usr/local/lib` (standard Homebrew installation).
- `/usr/local/Cellar/<package>/<version>/lib` (package-specific prefix of a standard Homebrew installation).
  - This one has an additional problem: The only shared library referenced in current HEAD is `libfuzzy` that is part of the `ssdeep` package (“formula” in Homebrew-lingo), but the `<package>` part is substituted with `*fuzzy*`, so this never yields anything.

My idea for a solution would be:

- When running `setup.py`, offer a switch `--library-search-prefix=` that would default to `/usr/local`.
- Put this into a constant in `src/binwalk/core/C.py`.
- Use that constant in `find_library` in the `Library` class instead of the various hard-coded paths. (This could even be extended to all non-Windows platforms, not just OS X.)

If this is an acceptable solution, I can try to come up with a pull request that implements this, though my Python skills are quite limited.

devttys0, 2015-06-01:
Thanks for the offer! The C library dependencies have always been a PITA, on all platforms. 

This is why there is no active code in the master branch that uses any C libraries; everything is now pure Python, so this should no longer be an issue. A stable release based on the current master branch is forthcoming.

Linux/OSX/Windows users can just run `setup.py install` to get up and running, though Python2 users are encouraged to install the `python-lzma` module (this module is now standard in Python3); see the [INSTALL.md](https://github.com/devttys0/binwalk/blob/master/INSTALL.md) file for more info.

closed 2015-06-01 by devttys0

UniqMartin, 2015-06-01:
Thanks for the quick response! I had failed to see that the only code referencing a shared library (`binwalk.modules.hashmatch.HashMatch` using `libfuzzy`) was no longer active …