ReFirmLabs/binwalk #142: Analyze .sig file type

manjesh23, 2015-07-14:
Hello Team,

Requesting you to help us analyze .sig firmware file type.

Regards,

Manjesh N
manjesh_n@hotmail.com

devttys0, 2015-07-16:
Do you have a link to the firmware image in question?

manjesh23, 2015-07-22:
Hello,

Its just a SonicOS, you can download from mysonicwall.com. If you still want me to attach the same, please let me know and I will be happy to do.

Waiting for your response.

Regards,

Manjesh N

devttys0, 2015-07-22:
Yes, please attach or provide a direct download link. Thanks!

manjesh23, 2015-07-23:
Hello,

Here is the path to download the .sig file

http://www.filedropper.com/swnsa-3600eng6222-19n

Regards,

Manjesh N

manjesh23, 2015-08-24:
Hello,

May we please have an update on this please?

Waiting for your response and thanks in advance.

Regards,

Manjesh

devttys0, 2015-08-25:
There's not much to analyze, unfortunately. There is some header data, but then around offset `0x400` the data gets very random. My guess would be that it's encrypted. Binwalk's entropy analysis shows some small, but regular, dips in the entropy, which might suggest a block cipher of some kind. 

Your best bet is probably to try to get on the hardware itself, which might give you some clues, or even an unencrypted image. But I think that just from this firmware update file, with no other information, options are very limited.

closed 2016-01-04 by devttys0