ReFirmLabs/binwalk #146: Error when running binwalk in init script

raddirad, 2015-08-05:
Hi,
I am using binwalk in a python script which is executed within an init script.
I got this error in version 1.0 in Ubuntu 14.04.
Now i compile binwalk from sources in hope to fix the error

running binwalk this way gives me the following error:

(4226)    [2] [Aug 05 08:49:09]      ==>>> Command '. /root/.profile ; /usr/local/bin/binwalk /mnt/hd/tmp/initrd.img > /mnt/hd/tmp/binwalk.out 
(4227)    ' failed (1):
(4228)    stdin: is not a tty
(4229)    Traceback (most recent call last):
(4230)      File "/usr/local/bin/binwalk", line 81, in <module>
(4231)        main()
(4232)      File "/usr/local/bin/binwalk", line 60, in main
(4233)        elif not modules.execute():
(4234)      File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 691, in execute
(4235)        obj = self.run(module)
(4236)      File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 707, in run
(4237)        obj = self.load(module, kwargs)
(4238)      File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 722, in load
(4239)        argv.update(self.dependencies(module, argv['enabled']))
(4240)      File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 748, in dependencies
(4241)        depobj = self.run(dependency.module, dependency=True, kwargs=dependency.kwargs)
(4242)      File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 707, in run
(4243)        obj = self.load(module, kwargs)
(4244)      File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 723, in load
(4245)        return module(**argv)
(4246)      File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 229, in __init__
(4247)        self.plugins = binwalk.core.plugin.Plugins(self)
(4248)      File "/usr/local/lib/python2.7/dist-packages/binwalk/core/plugin.py", line 93, in __init__
(4249)        self.settings = binwalk.core.settings.Settings()
(4250)      File "/usr/local/lib/python2.7/dist-packages/binwalk/core/settings.py", line 44, in __init__
(4251)        magic=self._magic_signature_files(user_only=True),
(4252)      File "/usr/local/lib/python2.7/dist-packages/binwalk/core/settings.py", line 69, in _magic_signature_files
(4253)        user_dir = os.path.join(self.user_dir, self.BINWALK_USER_DIR, self.BINWALK_MAGIC_DIR)
(4254)      File "/usr/lib/python2.7/posixpath.py", line 77, in join
(4255)        elif path == '' or path.endswith('/'):
(4256)    AttributeError: 'NoneType' object has no attribute 'endswith' (master.py|1512)

any suggestions?

devttys0, 2015-08-06:
I haven't tried running binwalk via an init script, but try the latest code from the master branch; there have been a lot of changes since the 2.0.1 release.

raddirad, 2015-08-07:
I tested it with 2.1 and I still get this annoying error.

Nevermind, I am currently writing the whole binwalk procedure in a shell script and executing it with 
su -s /bin/bash -c '/path/to/script' root
and it works

awesome tool by the way ;)

closed 2015-08-25 by devttys0