ReFirmLabs/binwalk #147: Winbond bootloader

plaes, 2015-08-13:
This was found via: https://github.com/devicenull/ipmi_firmware_tools

Apparently there are bunch of IPMI firmware blobs out there ( http://www.supermicro.com/products/motherboard/xeon/c202_c204/x9scl-f.cfm -> Select IPMI firmware from right), that cannot be extracted with binwalk:

```
$ binwalk SMT_X9_338.bin 
DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
59700         0xE934          Copyright string: " (c) Winbond Limited 2001 - 2006. All rights reserved.6. All rights reserved."
60835         0xEDA3          Copyright string: " 1995-1998 Mark Adler  termination"
1572864       0x180000        CramFS filesystem, little endian size 8372224 version #2 sorted_dirs CRC 0x6bfd8207, edition 0, 5183 blocks, 1101 files
9961472       0x980000        Zip archive data, at least v2.0 to extract, compressed size: 1124877,  uncompressed size: 2331112, name: "kernel.bin"
11086479      0xA92A8F        End of Zip archive
12058624      0xB80000        CramFS filesystem, little endian size 2445312 version #2 sorted_dirs CRC 0x34c60a53, edition 0, 1115 blocks, 218 files
```

The "Winbond Limited" thingy seems to be a bootloader for WPCM450 IMPI chip.

devttys0, 2015-08-13:
Could you be more specific as to what exactly binwalk isn't extracting? Both CramFS images and the ZIP archive extracted fine for me.

Based on an entropy analysis, there appears to be nothing else in the image except for the bootloader, which is just raw code starting at offset 0 in the firmware image (no compression, no header, etc):

```
$ binwalk -Y SMT_X9_338.bin 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             ARM executable code, 32-bit, little endian, at least 710 valid instructions
```

Actually the first four bytes are technically valid ARM, but are more likely a 4-byte signature or checksum. In any case, you can disassemble the bootloader if you just load the firmware image into a disassembler.

plaes, 2015-08-13:
ZIP file from `SMT_X9_338.bin` is broken for me:
```
$ unzip 980000.zip 
Archive:  980000.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of 980000.zip or
        980000.zip.zip, and cannot find 980000.zip.ZIP, period.
```

devttys0, 2015-08-13:
That's a common issue with firmware that have embedded ZIP files. The ZIP file is not broken, it just doesn't have an end-of-central-directory signature, which causes `unzip` to choke. (The `980000.zip` file is literally just a byte-for-byte copy of the data in the `SMT_X9_338.bin` image).

This is why binwalk uses `7z` for the extraction of ZIP files. If you have `7z` installed, `binwalk -e` will automatically invoke it and decompress the ZIP file.

plaes, 2015-08-13:
OK, my bad. This also explains where the `kernel.bin` file comes from:

9961472       0x980000        Zip archive data, at least v2.0 to extract, compressed size: 1124877,  uncompressed size: 2331112, name: "kernel.bin"

closed 2015-08-13 by plaes