ReFirmLabs/binwalk #150: Failed to Load Plugin Module MacOS 10.10.5

bl4ckic3, 2015-08-21:
I have problem with binwalk. 
Everytime I run it, I get an error for :
 binwalk firmware.mg 

WARNING: Failed to load plugin module 'compressd': Failed to locate libraries '['compress42']'

WARNING: Failed to load plugin module 'zlibvalid': Failed to locate libraries '['tinfl']'

WARNING: Failed to load plugin module 'compressd': Failed to locate libraries '['compress42']'

WARNING: Failed to load plugin module 'zlibvalid': Failed to locate libraries '['tinfl']'

Where I can add this libraries, previous issues solutions referring to copy dylib files which does not exist in the current version of binwalk.

niklaus520, 2015-08-22:
https://github.com/devttys0/binwalk/issues/77#issuecomment-77284013

just google it

bl4ckic3, 2015-08-23:
As I already mentioned I already did that.
 I manually installed binwalk: Result was the same. 
I tried to copy those files mentioned in the other issues  (e.g. copying *.dylib files to /usr/local/lib/ or 
 src/C/compress/libcompress42.dylib to /usr/local/lib/) but as I mentioned earlier such files does not exist in the Binwalk anymore.

niklaus520, 2015-08-23:
Have u tried compile the source code first? Cause i just download the source code, configure, make, make install then copy, all worked fine

bl4ckic3, 2015-08-23:
I downloaded the source from github. Here is the output from source installation (did it again right now) 

Majids-Air:binwalk-master majid$ sudo python setup.py install
Password:
running install
running build
running build_py
running build_scripts
running install_lib
copying build/lib/binwalk/__init__.py -> /usr/local/lib/python2.7/site-packages/binwalk
copying build/lib/binwalk/config/extract.conf -> /usr/local/lib/python2.7/site-packages/binwalk/config
copying build/lib/binwalk/core/__init__.py -> /usr/local/lib/python2.7/site-packages/binwalk/core
copying build/lib/binwalk/core/C.py -> /usr/local/lib/python2.7/site-packages/binwalk/core
copying build/lib/binwalk/core/common.py -> /usr/local/lib/python2.7/site-packages/binwalk/core
copying build/lib/binwalk/core/compat.py -> /usr/local/lib/python2.7/site-packages/binwalk/core
copying build/lib/binwalk/core/display.py -> /usr/local/lib/python2.7/site-packages/binwalk/core
copying build/lib/binwalk/core/idb.py -> /usr/local/lib/python2.7/site-packages/binwalk/core
copying build/lib/binwalk/core/magic.py -> /usr/local/lib/python2.7/site-packages/binwalk/core
copying build/lib/binwalk/core/module.py -> /usr/local/lib/python2.7/site-packages/binwalk/core
copying build/lib/binwalk/core/plugin.py -> /usr/local/lib/python2.7/site-packages/binwalk/core
copying build/lib/binwalk/core/settings.py -> /usr/local/lib/python2.7/site-packages/binwalk/core
copying build/lib/binwalk/magic/archives -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/magic/binarch -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/magic/bincast -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/magic/binwalk -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/magic/bootloaders -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/magic/code -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/magic/compressed -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/magic/console -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/magic/crypto -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/magic/ecos -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/magic/encoding -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/magic/executables -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/magic/filesystems -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/magic/firmware -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/magic/hashing -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/magic/images -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/magic/linux -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/magic/lzma -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/magic/misc -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/magic/network -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/magic/sql -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/magic/vxworks -> /usr/local/lib/python2.7/site-packages/binwalk/magic
copying build/lib/binwalk/modules/__init__.py -> /usr/local/lib/python2.7/site-packages/binwalk/modules
copying build/lib/binwalk/modules/binvis.py -> /usr/local/lib/python2.7/site-packages/binwalk/modules
copying build/lib/binwalk/modules/compression.py -> /usr/local/lib/python2.7/site-packages/binwalk/modules
copying build/lib/binwalk/modules/disasm.py -> /usr/local/lib/python2.7/site-packages/binwalk/modules
copying build/lib/binwalk/modules/entropy.py -> /usr/local/lib/python2.7/site-packages/binwalk/modules
copying build/lib/binwalk/modules/extractor.py -> /usr/local/lib/python2.7/site-packages/binwalk/modules
copying build/lib/binwalk/modules/general.py -> /usr/local/lib/python2.7/site-packages/binwalk/modules
copying build/lib/binwalk/modules/hashmatch.py -> /usr/local/lib/python2.7/site-packages/binwalk/modules
copying build/lib/binwalk/modules/heuristics.py -> /usr/local/lib/python2.7/site-packages/binwalk/modules
copying build/lib/binwalk/modules/hexdiff.py -> /usr/local/lib/python2.7/site-packages/binwalk/modules
copying build/lib/binwalk/modules/signature.py -> /usr/local/lib/python2.7/site-packages/binwalk/modules
copying build/lib/binwalk/plugins/arcadyan.py -> /usr/local/lib/python2.7/site-packages/binwalk/plugins
copying build/lib/binwalk/plugins/compressd.py -> /usr/local/lib/python2.7/site-packages/binwalk/plugins
copying build/lib/binwalk/plugins/cpio.py -> /usr/local/lib/python2.7/site-packages/binwalk/plugins
copying build/lib/binwalk/plugins/gzipextract.py -> /usr/local/lib/python2.7/site-packages/binwalk/plugins
copying build/lib/binwalk/plugins/gzipvalid.py -> /usr/local/lib/python2.7/site-packages/binwalk/plugins
copying build/lib/binwalk/plugins/jffs2valid.py -> /usr/local/lib/python2.7/site-packages/binwalk/plugins
copying build/lib/binwalk/plugins/lzmaextract.py -> /usr/local/lib/python2.7/site-packages/binwalk/plugins
[... cut ...]
266304        0x41040         LZMA compressed data, properties: 0x5D, dictionary size: 33554432 bytes, uncompressed size: 8495822 bytes
274723        0x43123         Zlib compressed data, compressed
278976        0x441C0         Zlib compressed data, best compression
316810        0x4D58A         Zlib compressed data, default compression
370708        0x5A814         Zlib compressed data, best compression
380113        0x5CCD1         Zlib compressed data, compressed
405030        0x62E26         Zlib compressed data, default compression
408535        0x63BD7         Zlib compressed data, default compression
..............

niklaus520, 2015-08-23:
The python installer failed to compile the lib, so just do:

```
./configure
make
sudo make install
sudo cp src/C/miniz/libtinfl.dylib /usr/local/lib/
sudo cp src/C/compress/libcompress42.dylib /usr/local/lib/
```

BTW, i suppose you haven't read all the comments in the page that i mentioned.

bl4ckic3, 2015-08-23:
I particularly mentioned what you quoted right now in my answer. Did you see my comment? 
Would you please tell me where do you find this libcompress42.dylib in the binwalk installation folder that I do not?

niklaus520, 2015-08-23:
[image]

bl4ckic3, 2015-08-23:
Which version of binwalk you are using. I do not find libs folder even in the current version of binwalk : 
https://github.com/devttys0/binwalk/tree/master/src/binwalk
[image]

niklaus520, 2015-08-23:
https://github.com/devttys0/binwalk/releases/tag/v2.0.1 

The release 2.0.1 version, i think it's the stable version

niklaus520, 2015-08-23:
I just clone the newest version code, i think they just removed configure file

bl4ckic3, 2015-08-23:
Fixed. Thanks. I forked it from github. In github source there were no lib and C folders.

devttys0, 2015-08-25:
Yes, the latest code in github has removed the C libraries, so this should no longer be a problem. A new stable release is probably overdue at this point. :) Thanks niklaus520!

closed 2015-08-25 by devttys0