ReFirmLabs/binwalk #151: Warnings during binwalk -A

alex-bender, 2015-09-07:
```
$ git clone https://github.com/devttys0/binwalk
$ cd  binwalk
$ sudo ./deps.sh
$ sudo python setup.py install
$ binwalk -A

WARNING: Error loading plugin 'compressd.py': /usr/local/lib/ctypes.so: undefined symbol: list_optarg
```

Warning repeated approx 20 times.

```
$ uname -a
Linux bender 3.19.0-15-generic #15-Ubuntu SMP Thu Apr 16 23:32:37 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

$ python -V 
Python 2.7.9

```

devttys0, 2015-09-10:
It sounds like an issue in `ctypes`, but I can't reproduce this error on my machine. The `compressd.py` plugin is not currently used though, so I've removed any code in `compressd.py` that could be responsible for invoking `ctypes`, which should fix this problem.

alex-bender, 2015-09-11:
I have uninstalled binwalk and the installed it again, but error is still present.

devttys0, 2015-09-27:
Nowhere does binwalk directly reference `list_optarg`, which leads me to believe there is something wrong with the `ctypes` Python interface. Can you start a Python shell and run `import ctypes` and `import ctypes.util` without errors/warnings?

alex-bender, 2015-09-28:
Here it is:

```
$ python
Python 2.7.9 (default, Apr  2 2015, 15:33:21) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import ctypes
>>> import ctypes.util
```

devttys0, 2015-10-01:
`compressd.py` in the latest master literally does nothing. See if this fixes your issue.

alex-bender, 2015-10-01:
Yes, it is fixed. Thanks
```
$ binwalk -A
$ echo $?
0
```

closed 2015-10-01 by alex-bender