ReFirmLabs/binwalk #152: How could I create a deflated file?

iddq, 2015-09-07:
Hello,

Can you please tell me how could I create a deflated file for testing the binwalk's -X function?

Thanks

iddq, 2015-09-07:
if this is a good testing method then there's some problem.

root@debian:/tmp/cs# cp /etc/passwd test
root@debian:/tmp/cs# gzip test
root@debian:/tmp/cs# gunzip -v -l test
method  crc     date  time           compressed        uncompressed  ratio uncompressed_name
defla db7872e8 Sep  8 00:06                 657                1704  62.8% test
root@debian:/tmp/cs# binwalk -X -e test.gz

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
650           0x28A           Raw deflate compression stream
652           0x28C           Raw deflate compression stream
653           0x28D           Raw deflate compression stream
655           0x28F           Raw deflate compression stream
656           0x290           Raw deflate compression stream

root@debian:/tmp/cs# ls -al _test.gz.extracted/
total 28
drwxr-xr-x 2 root root 4096 Sep  8 00:06 .
drwxr-xr-x 3 root root 4096 Sep  8 00:06 ..
-rw-r--r-- 1 root root    7 Sep  8 00:06 28A.deflate
-rw-r--r-- 1 root root    5 Sep  8 00:06 28C.deflate
-rw-r--r-- 1 root root    4 Sep  8 00:06 28D.deflate
-rw-r--r-- 1 root root    2 Sep  8 00:06 28F.deflate
-rw-r--r-- 1 root root    1 Sep  8 00:06 290.deflate

The extracted files are too short and invalid.

devttys0, 2015-09-10:
The raw compression module assumes the most common zlib config bits when attempting to detect deflated data. This means that it currently will detect some deflated data but not others. For example, it doesn't work on your .gz file, but I have other .gz files that it works fine against.

iddq, 2015-09-12:
https://github.com/iddq/tmp/blob/master/deflate-test.tgz

iddq, 2015-09-30:
Could you reproduce this bug?

content of package:
gzip binary
test script for reproducing the bug

devttys0, 2015-09-30:
Yes, this is a known issue.

```
The raw compression module assumes the most common zlib config bits 
when attempting to detect deflated data. This means that it currently will 
detect some deflated data but not others.
```

The solution is to iterate over all possible deflate configuration options, instead of just the most common, but that is currently not implemented. If you just need to test the zlib/deflate detection/extraction, you can use Python's zlib module to create compressed data for testing.

devttys0, 2015-12-14:
Turns out, the Python `zlib` module already handles full raw decompression. This is fixed in the latest master commit, it finds and extracts your gzip data just fine now. :)

closed 2015-12-23 by devttys0

stokito, 2020-06-08:
here is an example of raw DEFLATE compression and decompression https://github.com/stokito/deflate