ReFirmLabs/binwalk #153: Carved files are often equal in size to the original file

RomanHargrave, 2015-09-10:
I have experienced this a couple times.

I will use binwalk to analyze a memory image or unknown data blob. When it carves files, the files are often close to the file from which they were carved in size, even if there are so many valid detections that is is simply not possible for all carved data to fit in the original file.

For example, I have a file -- `x` that is 1.3 GB. `x` contains several contiguous zlib-compressed blobs. Binwalk will carve out _that many_ ***unextracted*** blobs, but all between 1.0 GB and 1.3 GB.

devttys0, 2015-09-10:
If binwalk can detect the size of the file it is carving, then it will only carve that amount of data from file `x`. This generally requires that the file format being carved has some size field (typically in the header) that can be parsed out by the binwalk signature.

As the zlib file format has no such identifying data (you have to parse all the zlib data until you find the end of the compressed data), binwalk simply carves everything from the start of the detected file to the end of file `x`. zlib is not unique in this respect, and adding such parsers for every file type that fits into this category is impractical. Some file types (e.g., JFFS2) have *no* way of detecting the end of the file.

While it would be possible to say 'only extract zlib data up to the next valid signature', this is problematic as there is no guarantee that the next "valid" signature is actually valid, hence doing so could truncate the zlib data stream, causing extraction utilities to barf.

Obviously carving many very large files can eat up disk space very quickly, which is undesirable. The easiest way to work around this is to run binwalk with the `--extract` and `--rm` options. This will cause binwalk to carve and extract the zlib data, and subsequently delete original carved file, leaving you only with the actual data. You can also limit the max carved file size with `--size`.

The only other alternative would be to add many specific cases for zlib and similar file types, which gets very messy very quickly and is difficult to do generically. If this is an issue that you commonly encounter in the files that you work with, it would be much easier to write a custom script using the Python API to handle your specific case.

RomanHargrave, 2015-09-10:
I can understand that, but I have had situations where zlib blocks from near the end of the file are larger than the space between their beginning and the end of the file.

RomanHargrave, 2015-09-10:
woops

devttys0, 2015-09-27:
Hrm, yes, that shouldn't happen! I'll see if I can reproduce the issue.

emkay77, 2015-10-16:
I have the same problem - and the -r or --rm Option doesn't help either because it won't cleanup some 'intermediate' Files.
I have a Routerfirmware with JFFS2-Parts. Binwalk converts them from big-endian to little-endian but the 'auto-cleanup' won't delete the '.jffs2.le'-Files...

devttys0, 2015-12-16:
@RomanHargrave, I haven't been able to reproduce this issue, either with zlib files or any other type of file. Since your first comment and subsequent comment(s) seem (to me) to be stating slightly different issues, let me make sure I understand exactly what is happening.

In the initial issue seemed to be that the sum of the sizes of the extracted zlib files was larger than the original input file. This is expected behavior as binwalk doesn't know where a zlib data should end, so it just extracts everything from the beginning of the zlib header to the end of the input file. If, for example, you had two zlib headers inside file `x`, one at the very beginning of file `x` and another exactly half way through file `x`, then you would get two extracted files: the first would be the exact size of file `x`, the second would be exactly half the size of file `x`. The sum of these file's sizes would obviously be larger than the total size of file `x`.

In a subsequent comment you said that zlib blocks from near the end of the file are larger than the space between their beginning and the end of the file. So, for example, if there were zlib data 100 bytes away from the end of file `x`, you might get an extracted file whose size was larger than 100 bytes (say, 200 bytes for example). I have not been able to reproduce this. Just to be sure, you aren't looking at the size of the uncompressed zlib file, are you?

@emkay77, the specific jffs2 issue should no longer be an issue in the latest code, as it uses a different jffs2 extraction utility.

malaterre, 2016-09-29:
I am also experiencing this over here [as well](http://reverseengineering.stackexchange.com/questions/13616/simple-carving-of-zip-file-using-binwalk).

Would be be possible to either clearly document that in the help: "carved file may contain all data up to the end of the original file".

As a side note, some file do not strictly contains length but do contains marker for the end (PDF, JPEG, gzip). So it should be possible to carve using both a header (file signature) and a 'trailer' (eg. FFD9 for JPEG).

malaterre, 2016-09-29:
Just for completeness, here is a way to create one such file:

```
$ dd if=/dev/zero of=head bs=1 count=512
$ dd if=/dev/zero of=junk bs=1 count=512000
$ wget https://www.debian.org/logos/openlogo-100.jpg
$ cat head binwalk-master.zip junk > full
$ binwalk -z -C demo -D 'jpeg:jpg' full
$ ls -al demo/_full.extracted/200.jpg openlogo-100.jpg 
-rw-r--r-- 1 user user 520420 Sep 29 11:32 demo/_full.extracted/200.jpg
-rw-r--r-- 1 user user   8420 Dec 11  1999 openlogo-100.jpg
```

malaterre, 2016-11-27:
For anyone reading this, you need to use a recent `p7zip` implementation. binwalk expect the `unzip` implementation to read a file from the start, while some are reading file starting from the end. Leading to odd behavior. See for more info [here](http://reverseengineering.stackexchange.com/questions/13944/automatically-extract-known-file-types-using-binwalk).

closed 2018-04-12 by CoffeeExpress

mefistotelis, 2025-01-05:
If the header does not contain any indication of size, `binwalk` could still speculate on the size based on existing fields. ie. GIF image of 8 bit and resolution 1000x1000 will not exceed `header_sz + 1000*1000*1.2` even in a worst case in terms of compression. Other formats sometimes have offsets to something near the end, even if not the end per se.