ReFirmLabs/binwalk #161: binwalk makes spurious reads from standard input

fstirlitz, 2015-10-25:
I often find myself unable to use my pager while running `binwalk -W`. I often have to hit a key at least twice before the pager catches it.

Running `binwalk < /dev/null` prevents any output from appearing. Running `binwalk </dev/zero` exhausts my memory in about two seconds. Since binwalk isn't supposed to read anything from standard input unless explicitly requested, this is behaviour is nonsensical.

devttys0, 2015-10-27:
The binwalk script does [read from stdin](https://github.com/devttys0/binwalk/blob/master/src/scripts/binwalk#L30).

fstirlitz, 2015-10-30:
Well, remove it then.

devttys0, 2015-11-20:
It's a very useful feature. If it's a PITA for you, there are a couple of options available. First, you could just output the data to a file, then feed the file to your pager:

`
$ binwalk -W somefile.bin > /tmp/hex && more /tmp/hex
`

Since no tool can be everything to everyone, binwalk is customizable and scriptable. This python script gives you all the features of binwalk, but without the display status feature:

```
import binwalk
binwalk.scan()
```

The above script will accept all the same command line options as the packaged binwalk script.

fstirlitz, 2015-11-21:
Sure, I can even re-write binwalk from scratch. Why not?

Breaking functionality that works just fine in literally every other command-line utility is a misfeature. I haven't ever used this supposedly "very useful feature", I even had no idea it existed until I looked into the source code and winced at how poorly it was implemented. No other command-line utility behaves like this, and it's not even documented anywhere. As such it violates the Rule of Least Surprise, a few times over.

For what it's worth, GNU dd outputs current progress on standard error when sent, depending on the platform, either SIGINFO or SIGUSR1. I used it maybe twice in my life: it's hardly a killer app.

devttys0, 2015-11-23:
```Sure, I can even re-write binwalk from scratch. Why not?```

Point taken, but writing one or two lines of bash/python and re-writing an entire tool are *slightly* different, wouldn't you say? :) The advantage of OSS is that users can add/remove features, or fix broken ones, *without* having to re-write the whole tool, and even submit those changes back to the original project to benefit everyone.

```it's not even documented anywhere. As such it violates the Rule of Least Surprise```

Excellent points both; at the very least it should have at least been documented, there's no excuse for that.

```I haven't ever used this supposedly "very useful feature"```

I'm not sure how this is relevant, but it doesn't seem like a logical argument for the complete removal of a feature. 

```For what it's worth, GNU dd outputs current progress on standard error when sent, depending on the platform, either SIGINFO or SIGUSR1.```

I like the idea of using signals like dd does, but it wouldn't work well across platforms (i.e., Windows). I think the obvious solution is to just make the status display a command line option. Output could be sent to a user-specified temp file or unix/network socket so it doesn't clutter the screen, and no dynamic user input would be required. 

Users who always want the status feature enabled (usually you only want to see the status if the command is taking a long time, and you often don't know that it will take a long time until after you've already run the command) could set up an alias in their shell profile, or it could be made a configuration option.

devttys0, 2015-11-23:
The latest code in master no longer requires input from stdin for status display.

fstirlitz, 2015-11-24:
That's better. Thank you.

closed 2015-11-25 by devttys0