ReFirmLabs/binwalk #163: binwalk cannot properly identify and extract firmware contents

rodrigo-menezes, 2015-11-19:
Here's the output I get running binwalk on an Intelbras WRN 300 router [firmware](http://www.intelbras.com.br/sites/default/files/downloads/firmware_wrn300_versao_1.0.rar) file:

```
DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             TRX firmware header, little endian, header size: 28 bytes, image size: 1015808 bytes, CRC32: 0x7D8B525E flags: 0x0, version: 1
28            0x1C            LZMA compressed data, properties: 0x5D, dictionary size: 65536 bytes, uncompressed size: 2740652 bytes
```

Then, when running binwalk on the file created from decompressing the LZMA section:

```
DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
120           0x78            eCos kernel exception handler, architecture: MIPSEL, exception vector table base address: 0x80000200
256           0x100           eCos kernel exception handler, architecture: MIPSEL, exception vector table base address: 0x80000200
1796124       0x1B681C        eCos RTOS string reference: "ecos_name"
1796136       0x1B6828        eCos RTOS string reference: "ecos"
1797553       0x1B6DB1        eCos RTOS string reference: "eCos Router/AP %s "
1798864       0x1B72D0        Unix home path string: "/home/share/work/N30_ITB01/N30_ITB01/src/router/shared/nvparse."
1904924       0x1D111C        Unix home path string: "/home/share/work/N30_ITB01/N30_ITB01/src/router/nas/nas_wksp.c"
1905236       0x1D1254        Unix home path string: "/home/share/work/N30_ITB01/N30_ITB01/src/router/nas/nas_wksp_ra"
1932588       0x1D7D2C        Unix home path string: "/home/share/work/N30_ITB01/N30_ITB01/src/../src/router/libbcm/b"
1932736       0x1D7DC0        Unix home path string: "/home/share/work/N30_ITB01/N30_ITB01/src/../src/wps/common/shar"
1934688       0x1D8560        eCos RTOS string reference: "ECOS"
1936792       0x1D8D98        XML document, version: "1.0"
1937612       0x1D90CC        XML document, version: "1.0"
1940004       0x1D9A24        HTML document header
[...]
2678936       0x28E098        HTML document header
2697928       0x292AC8        HTML document footer
2697940       0x292AD4        GIF image data, version "89a", 976 x 2
```

The program fails to properly extract this image's contents. Extracted XML and HTML files contain "garbage", most likely parts of other (binary) files in the image.

**Note**: The WRN 300 appears to be a rebranded Tenda N30. It is possible to get firmware files for that Tenda router [here](http://www.tendacn.com/en/services/downlist-404.html). They are similar to the WRN 300 firmware file, and the same issue happens when trying to extract them.

devttys0, 2015-11-20:
The only issue is that binwalk doesn't know how big the HTML or XML files are when it extracts them, so it just extracts all the content from the beginning of the HTML/XML header to the end of the file.

If you look inside the extracted files, you'll see that they do in fact begin with valid HTML/XML data, and after that data there is a bunch of trailing "junk". Currently, there is no way to cleanly handle this in binwalk, but carving HTML files isn't really the main purpose of binwalk, and there are existing tools that handle this type of thing very well.

As binwalk's output above indicates, you're dealing with a Real Time Operating System, specifically, eCos. Most RTOS's are just one big lump of code and data; they typically don't have a proper file system, nor do they have a concept of user vs kernel space as you do in Linux. It's literally just a big blob of code concatenated with a big blob of data.

IMHO, binwalk has performed its job of identifying and extracting the firmware for you: what you have can now be fed to a disassembler, or to other data/code processing tools for further analysis. If you run `binwalk -A`, it will even tell you what architecture the code is, so that you know which disassembler to use.

Note that the HTML and XML documents probably do have some custom data structure associated with them (or maybe even something standard that is specific to eCos?); if this structure can be identified and reverse engineered, then signatures and extraction capabilities could be added without much trouble. But this would obviously require additional RE of the firmware.

rodrigo-menezes, 2015-11-20:
Yes, the files do begin with valid HTML/XML data. I understand that carving those files isn't the main purpose of binwalk and that it might be hard to do properly without knowledge of the data structures used.

But I was wondering: how exactly binwalk determines the end of an HTML/XML file in those cases? Couldn't there be some improvement for the method it uses? For instance, if non-character data starts appearing in the middle of an HTML document, I believe it would be pretty safe to assume that the file has ended.

If this is not adequate considering binwalk's purpose, or would create too many problems, please close this issue.

devttys0, 2015-11-22:
The problem is that binwalk doesn't detect the end of HTML or XML files, or for that matter, any file which does not specify its size in a header field. That's why the extracted HTML/XML files are copied from the start of the HTML/XML tag to the end of the firmware file.

Binwalk actually used to perform end of file detection for some file types such as this, but it was quite hacky and given that better tools exist specifically for file carving, it was removed in the 2.0 re-write. Check out scalpel: https://github.com/sleuthkit/scalpel

closed 2015-11-22 by devttys0