ReFirmLabs/binwalk #164: Does not run on Windows 7.

farmdve, 2015-11-20:
Whether I build it from master or use `pip install binwalk`, the result is the same.

```
C:\Python271\Scripts>python binwalk.py
Traceback (most recent call last):
  File "binwalk.py", line 23, in <module>
    import binwalk
  File "C:\Python271\Scripts\binwalk.py", line 24, in <module>
    import binwalk.modules
ImportError: No module named modules
```

Also, the file created in the Scripts directory is without an extension.

chayankhoidream, 2015-11-21:
Use Oracle VM VirtualBox virtual machine app, find ubuntu and install on a VM, then install on ubuntu instance

devttys0, 2015-11-22:
The build from master works for me with Python2.7 on Win7x64. 

I didn't create the PyPi repository, and it does not appear to be maintained. I suspect what has happened is you first attempted to install via `pip`, then after that failed, you built it from master. If so, then the `pip` install probably has some conflicting files that it installed which are causing the issue.

Be sure to completely delete the binwalk module (should be in `C:\Python271\lib\site-packages`) and re-build from master.

With that said, some binwalk features (especially file extraction) are limited on Windows, so chayankhoidream's suggestion of running it in a Linux VM will generally be a better solution than running it natively in Windows.

wittrup, 2015-12-07:
Maybe not relevant... Binwalk works for me with Python3.5 on Win10x64 trough MINGW64 (coming with GIT-SCM installer).  Binwalk pip installed.

closed 2016-01-04 by devttys0

eladkarako, 2018-07-02:
@farmdve The cause is <code>binwalk.py</code> way of discovering the <code>lib</code> folder, 
(up once and down once to <code>lib</code> folder).

It often breaks on Windows but this wrapper seems to work well so far:
https://github.com/eladkarako/mods/tree/master/binwalk/

I've built it on Cygwin and modified it slightly with a batch file, 
to make it easier to run on Windows (no build needed).

I'm using WinPython, 
with <code>C:\Python\python-3.6.3</code> added to the system-Path.

It would probably make other Windows-user's life easier too..

mirh, 2018-07-03:
>python C:\Python36\Scripts\binwalk 

Seems enough to get everything to work.