ReFirmLabs/binwalk #165: python3 regression in statuserver.py: ImportError: No module named 'SocketServer'

anthraxx, 2015-11-25:
Hey, you newest changes introduced a regression related to python3

https://github.com/devttys0/binwalk/blob/596489cc8880271dceed603cd9bd7a9217685976/src/binwalk/core/statuserver.py#L6

    
    ImportError: No module named 'SocketServer'
    
The problem is that this module is handled/named differently in python2 and python3 (SocketServer vs socketserver). It would be awesome if you could provide compatibility to both by catching that ImportError and doing a "switch"

devttys0, 2015-11-25:
Sorry, should have done Python3 tests before pushing! Should be fixed now.

anthraxx, 2015-11-25:
@devttys0 damn you're fast! just finished and wanted to push my fix to create a pull-request :stuck_out_tongue_closed_eyes: 

Verified your fix and its working fine :+1: 
PS: sweet feature, awesome for very large files :heart_eyes:

anthraxx, 2015-11-25:
PS: any plans for 2.1.0? maybe as a xmas gift? :stuck_out_tongue_winking_eye:

closed 2015-11-25 by anthraxx

devttys0, 2015-11-27:
```damn you're fast!```

Well, you caught me at a good time. :)

```any plans for 2.1.0? maybe as a xmas gift?```

I'll see what I can do. ;)