ReFirmLabs/binwalk #166: no attribute 'MAGIC_NO_CHECK_TEXT'

metemaddar, 2015-11-26:
I was trying to work with a bin file using Firmware Modification Kit.
I got the error:
Firmware Mod Kit (extract) 0.99, (c)2011-2013 Craig Heffner, Jeremy Collake

Preparing tools ...
Scanning firmware...
Traceback (most recent call last):
  File "./src/binwalk-1.0/src/bin/binwalk-script", line 5, in <module>
    import binwalk
  File "/home/flight/Downloads/fmk/src/binwalk-1.0/src/bin/binwalk/__init__.py", line 12, in <module>
    class Binwalk:
  File "/home/flight/Downloads/fmk/src/binwalk-1.0/src/bin/binwalk/__init__.py", line 28, in Binwalk
    DEFAULT_FLAGS = magic.MAGIC_NO_CHECK_TEXT | magic.MAGIC_NO_CHECK_ENCODING | magic.MAGIC_NO_CHECK_APPTYPE | magic.MAGIC_NO_CHECK_TOKENS
AttributeError: 'module' object has no attribute 'MAGIC_NO_CHECK_TEXT'
sort: cannot read: /home/flight/Downloads/fmk/fmk/logs/binwalk.log: No such file or directory
Extracting 0 bytes of  header image at offset 0
ERROR: No supported file system found! Aborting...

File is this:
http://www.sat2hd.tv/attachment.php?s=6f84153a1a0d918d1ce0c8c9f22cb5eb&attachmentid=35751&d=1344260427

Thanks a lot

devttys0, 2015-11-27:
This error is usually due to an old or incompatible python-magic module. The binwalk bundled with FMK is older, and this issue no longer pertains to the main binwalk code anymore, as the python-magic module is no longer used.

You should either install the official python-magic module, included in the file utility source code, or install the latest binwalk and replace `binwalk-script` with a symlink to the latest binwalk.

However, in this case, FMK will not do you much good. The firmware image you have linked to appears to be running the Nucleus PLUS RTOS, and there is no file system as such. Further, the firmware image is nibble swapped, that is, every 2 bytes are swapped, so by default binwalk won't find anything.

The following command will byte-swap, then scan, the firmware image, and should recover and decompress two zlib compressed ELF files:

```$ binwalk -g 2 -e 'HIVION-HV-9191X-fast USB MAX29LV160T.bin'```

These two ELF files contain MIPS executable code, appear to be the main core of the firmware, and are complete with symbol tables. You can disassemble them with any MIPS ELF disassembler.

closed 2015-12-23 by devttys0

maxi1192, 2016-11-04:
For me this does not solve the problem

ArcAIN6, 2017-07-13:
same issue here, new install of Kali with binwalk, and FMK, and it's still present.