ReFirmLabs/binwalk #168: binwalk Fails if trying to analyse file inside folder with special chars (ie é | à)

rebrec, 2015-11-28:
It won't work if target file is in : ~/T**é**l**é**chargements/chroot/ 
It will work if target file is in :     ~/T**e**l**e**chargements/chroot/ 

``` bash
> # binwalk -eM upgrade-2.4.0.bin 

Scan Time:     2015-11-28 16:59:12
Target File:   /home/rebrec/Téléchargements/chroot/upgrade-2.4.0.bin
MD5 Checksum:  52ba767db3780a7af5b915e45d56950a
Signatures:    344

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------

Signature Exception: 'ascii' codec can't decode byte 0xc3 in position 55: ordinal not in range(128)
----------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 565, in main
    retval = self.run()
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/signature.py", line 177, in run
    self.scan_file(fp)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/signature.py", line 160, in scan_file
    self.result(r=r)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 454, in result
    getattr(self, dependency.attribute).callback(r)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/extractor.py", line 165, in callback
    (extraction_directory, dd_file, scan_extracted_files) = self.extract(r.offset, r.description, r.file.path, size, r.name)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/extractor.py", line 450, in extract
    fname = self._dd(file_path, offset, size, rule['extension'], output_file_name=name)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/extractor.py", line 632, in _dd
    binwalk.core.common.debug("Carved data block 0x%X - 0x%X from '%s' to '%s'" % (offset, offset+size, file_name, fname))
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 55: ordinal not in range(128)
----------------------------------------------------------------------------------------------------
```

devttys0, 2015-11-28:
Looking into a patch for this. A work around for the moment is to use Python3 instead of Python2, which should handle unicode paths "automagically".

rebrec, 2015-11-29:
Thanks for the workaroud

devttys0, 2015-11-30:
This was the result of a recent merge, which fixed one problem, but introduced another. You should no longer encounter this issue with the latest code from the master branch.

closed 2015-12-23 by devttys0