ReFirmLabs/binwalk #173: binwalk can't identify firmware signature

sh3dow, 2015-12-24:
i have Huawei router and binwalk can't identify firmware signature
also most of extracted file is zlib 

hexdump -C
``` 00000050  00 00 00 00 00 00 00 00  00 00 00 00 55 aa 5a a5  |............U.Z.|
00000060  64 00 00 00 01 00 00 00  48 57 45 57 31 31 2e 31  |d.......HWEW11.1|
00000070  00 01 00 00 71 01 00 00  32 30 31 34 2e 30 32 2e  |....q...2014.02.|
00000080  32 30 00 00 00 00 00 00  32 33 2e 32 37 2e 32 33  |20......23.27.23|
00000090  00 00 00 00 00 00 00 00  3a 4d 50 57 55 44 50 3a  |........:MPWUDP:|
000000a0  56 32 30 30 52 30 30 31  43 31 31 35 53 50 31 30  |V200R001C115SP10|
000000b0  30 00 00 00 00 00 00 00  22 03 00 10 00 00 1c 1e  |0.......".......|
000000c0  1f 8b 08 00 00 00 00 00  00 0b 2b 08 49 4c ca 49  |..........+.IL.I|
000000d0  f5 48 4d 4c 61 00 81 06  27 7f ff 90 20 7f df f8  |.HMLa...'... ...|
000000e0  30 03 43 3d 03 23 06 06  0f 33 4b 23 03 e7 e0 f8  |0.C=.#...3K#....|
000000f0  50 97 00 b0 02 a7 fc fc  12 9f 7c a8 6a 14 c0 02  |P.........|.j...|
00000100  c4 7f f4 41 98 91 11 21  ea 57 e6 94 98 9c ed 13  |...A...!.W......|
00000110  e2 8a a2 0e 04 24 e0 22  bc 20 f5 1c c8 ea dd 43  |.....$.". .....C|
00000120  11 26 c8 60 a8 e7 43 52  0f 72 4f 50 7e 2e 92 3b  |.&.`..CR.rOP~..;|
00000130  4c 80 d8 06 a8 c2 81 b1  c1 9e c1 80 c1 81 c1 80  |L...............|
00000140  09 c9 3d 98 ea 4b 80 b2  26 40 6c 81 55 7d 58 45  |..=..K..&@l.U}XE|
00000150  78 7e 51 76 31 92 fa 35  4c 0c 0c bf 40 2a 98 60  |x~Qv1..5L...@*.`|
00000160  ea 99 91 d4 fb e4 a7 e7  a3 86 cb 1a 16 06 34 20  |..............4 |
00000170  80 a4 5e bf 32 31 2d ad  d8 00 8b fa 03 70 11 36  |..^.21-......p.6|
```
binwalk output
``` 
DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
192           0xC0            gzip compressed data, from NTFS filesystem (NT), NULL date (1970-01-01 00:00:00)
678           0x2A6           gzip compressed data, from NTFS filesystem (NT), NULL date (1970-01-01 00:00:00)
33226         0x81CA          gzip compressed data, from NTFS filesystem (NT), NULL date (1970-01-01 00:00:00)
708934        0xAD146         gzip compressed data, from NTFS filesystem (NT), NULL date (1970-01-01 00:00:00)
9186686       0x8C2D7E        gzip compressed data, from NTFS filesystem (NT), NULL date (1970-01-01 00:00:00)
9218404       0x8CA964        gzip compressed data, from NTFS filesystem (NT), NULL date (1970-01-01 00:00:00)
12564378      0xBFB79A        gzip compressed data, from NTFS filesystem (NT), NULL date (1970-01-01 00:00:00)
29549318      0x1C2E306       gzip compressed data, from NTFS filesystem (NT), NULL date (1970-01-01 00:00:00)
39067308      0x2541EAC       gzip compressed data, from NTFS filesystem (NT), NULL date (1970-01-01 00:00:00)
``` 
i upload the file 
http://www23.zippyshare.com/v/LaEx3OlZ/file.html

devttys0, 2015-12-26:
I get the same binwalk output, all of the gzip files extract properly for me, and they all appear to contain valid data (several are YAFFS file systems). 

Can you elaborate on what you mean by `all of extracted files is zlib and empty folders`? Are the gzip files not extracting properly? Or were you expecting binwalk to find something else?

sh3dow, 2015-12-30:
First time I use `-Me` so binwalk extract all file, most of files zlib and empty folders,The next time I use only `-e` and extract file one by one, most of files is `elf`,`xml` and `crt`. I was looking for folders inside `filesystem` like `bin  dev  etc  init  jffs  lib  mnt  proc  sbin  sys  tmp  usr  var` didn't find anything maybe they not using busybox. also big `xml`  file have many `elf` files inside, so this firmware not busybox based.
last thing i do is using `-D` for extract only `YAFFS filesystems` the command i use `binwalk -D 'filesystem:d' 1C2E306` 
`ls |grep .d`
0.d
3030.d
748410.d
7C6BF0.d
also `-D` command not work well, the command extract YAFFS filesystems with all files inside `1C2E306` but YAFFS filesystems have extension `.d`
i extract  `0.d` and also have `YAFFS ` so i use `-D` , put it like loop every YAFFS  file i find inside it many YAFFS  files
---
if you can analyze the firmware and tell me what you find that will help me a lot.

sh3dow, 2015-12-30:
First time I use `-Me` so binwalk extract all file, most of files zlib and empty folders,The next time I use only `-e` and extract file one by one, most of files is `elf`,`xml` and `crt`. I was looking for folders inside `filesystem` like `bin  dev  etc  init  jffs  lib  mnt  proc  sbin  sys  tmp  usr  var` didn't find anything maybe they not using busybox. also big `xml`  file have many `elf` files inside, so this firmware not busybox based.
last thing i do is using `-D` for extract only `YAFFS filesystems` the command i use `binwalk -D 'filesystem:d' 1C2E306` 
`ls |grep .d`
0.d
3030.d
748410.d
7C6BF0.d
also `-D` command not work well, the command extract YAFFS filesystems with all files inside `1C2E306` but YAFFS filesystems have extension `.d`
i extract  `0.d` and also have `YAFFS ` so i use `-D` , put it like loop every YAFFS  file i find inside it many YAFFS  files

if you can analyze the firmware and tell me what you find that will help me a lot.

devttys0, 2015-12-30:
```I was looking for folders inside filesystem like bin dev etc init jffs lib mnt proc sbin sys tmp usr var didn't find anything maybe they not using busybox.```

If you run `strings` on the files that binwalk did extract, you'll find many references to `busybox`, so it's a pretty good bet that this firmware is `busybox` based.

```also big xml file have many elf files inside, so this firmware not busybox based.```

Since XML/HTML files have no header information regarding their file size, binwalk does not know how big the XML/HTML data is when it extracts it. Therefore, binwalk will extract *all* data from the start of the XML file to the end of the firmware image. If you want to carve XML/HTML data, there are already better tools for specifically doing that.

```also -D command not work well, the command extract YAFFS filesystems with all files inside 1C2E306 but YAFFS filesystems have extension .d```

The `-D 'filesystem:d'` command you used told binwalk to do exactly that. Since your `-D` command did not specify an extraction utility to run, no other actions were performed on the `.d` files.

Binwalk does not know how to extract YAFFS file systems. Unfortunately, I have not yet found any good extraction utilities for YAFFS (some exist, such as `yaffs2utils` and `unyaffs` but I've never gotten them to actually extract a YAFFS file system). Furthermore, last I checked, the Linux kernel did not have inherent support for mounting YAFFS file systems either.

sh3dow, 2015-12-31:
I appreciate your help
Thank you
`Unfortunately, I have not yet found any good extraction utilities for YAFFS`
I hope you find way to extract YAFFS.

devttys0, 2016-01-01:
Yes, unfortunately the existing `unyaffs` and `yaffs2utils` tools wouldn't process the YAFFS file systems in this firmware, even when given the correct page size and spare size values. Not sure why.

I wrote a [crude YAFFS extraction tool](https://github.com/devttys0/yaffshiv). It currently does not preserve file permissions or ownership, and does not support extracting special device files, but it does extract normal files, directories, and symlinks, and works on the YAFFS file systems in your firmware image.

You need to know the correct page size and spare size that the YAFFS image was created with, which for your firmware are `4096` and `16`, respectively:

```bash
$ ./yaffshiv.py 4096 16 BFB79A fs1
Created 30 directories, 276 files, and 110 symlinks.

$ ./yaffshiv.py 4096 16 1C2E306 fs2
Created 24 directories, 400 files, and 0 symlinks.
```

This should at least give you some files to work with!

devttys0, 2016-01-04:
The latest binwalk commit adds support for automatic extraction of YAFFS file systems using `yaffshiv` (you have to download and install `yaffshiv`, of course). Tested and working against your provided firmware image.

closed 2016-01-04 by devttys0

AliRazaManzoor, 2019-03-31:
> Yes, unfortunately the existing `unyaffs` and `yaffs2utils` tools wouldn't process the YAFFS file systems in this firmware, even when given the correct page size and spare size values. Not sure why.
> 
> I wrote a [crude YAFFS extraction tool](https://github.com/devttys0/yaffshiv). It currently does not preserve file permissions or ownership, and does not support extracting special device files, but it does extract normal files, directories, and symlinks, and works on the YAFFS file systems in your firmware image.
> 
> You need to know the correct page size and spare size that the YAFFS image was created with, which for your firmware are `4096` and `16`, respectively:
> 
> ```shell
> $ ./yaffshiv.py 4096 16 BFB79A fs1
> Created 30 directories, 276 files, and 110 symlinks.
> 
> $ ./yaffshiv.py 4096 16 1C2E306 fs2
> Created 24 directories, 400 files, and 0 symlinks.
> ```
> 
> This should at least give you some files to work with!

thats right for unpacking yaffs file sysem ,i checked it worked 

> debian@debian:~/Desktop/ls/a$ yaffshiv -a -f 05-00000107-System_R1 -d ali
Found 136 file objects with the following YAFFS settings:
Page size: 2048
Spare size: 16
ECC layout: True
Endianess: Little

Extracting file objects...
Created 13 directories, 69 files, and 53 links.

Question is How to repack it back .