ReFirmLabs/binwalk #174: please clarify new dependencies

LocutusOfBorg, 2015-12-24:
Hi, I see the new release has dropped the shared libraries and much stuff.

While this is nice for Debian packaging, I see however many new external repositories.
github.com/devttys0/sasquatch
github.com/sviehb/jefferson
my.smithmicro.com/downloads/files/stuffit520.611linux-i386.tar.gz
github.com/jrspruitt/ubi_reader

and maybe others...

should I package them all?

devttys0, 2015-12-26:
Yes, those new extractors, as well as [Sleuth Kit](https://github.com/sleuthkit/sleuthkit) have been added as external extractors (though I think `Sleuth Kit` is already packaged for Debian).

If you want to package them that would be awesome, but there might be some issues with some of them. For instance, `sasquatch` is just a hacked up version of `unsquashfs`, and isn't kept up to date with all the latest `unsquashfs` patches/releases.

LocutusOfBorg, 2015-12-28:
ok wonderful.

So I added
``` cramfsprogs, cramfsswap, openjdk-9-jdk, sleuthkit, squashfs-tools,```

I'll upload shortly in Debian..

Sasquatch is a no-go, I don't like duplicates (maybe consider patching unsquashfs if it doesn't meet your needs!)

thanks!

G.

closed 2015-12-28 by LocutusOfBorg

devttys0, 2015-12-28:
`openjdk-9-jdk` shouldn't be required; binwalk previously used some java tools to extract certain files, but `7z` is now used instead.

LocutusOfBorg, 2015-12-28:
ok thanks!
actually the dependency was
openjdk-7-jdk | openjdk-8-jdk | openjdk-9-jdk
but I dropped all of them according to your previous comment