ReFirmLabs/binwalk #177: Fedora packaging recommendations

swt2c, 2016-01-04:
Hi, I'm the Fedora package maintainer, and I'm looking to update the Fedora package to 2.1.1 (thanks for your work!).  I see the C code is gone but there are a number of new optional dependencies.  Any recommendation as to which of these I should flag as *required* vs *optional* (Fedora now has weak dependencies)?  If I flag them all as optional, would I lose any functionality as compared to version 2.0.0?

swt2c, 2016-01-04:
I went through the list, it turns out that most but not all of them are in Fedora already:
pyqtgraph -> in Fedora, already an optional dep
capstone -> in Fedora
mtd-utils -> in Fedora
gzip -> in Fedora
bzip2 -> in Fedora
tar -> in Fedora
arj -> in Fedora
lhasa -> not in Fedora
p7zip -> in Fedora
cabextract -> in Fedora
cramfsprogs -> not in Fedora
cramfsswap -> not in Fedora
squashfs-tools -> in Fedora
sasquatch -> in Fedora
jefferson -> not in Fedora
ubi_reader -> not in Fedora
unstuff -> not in Fedora (non-free)
sleuthkit -> in Fedora

I'm leaning towards flagging them all as optional, but would like your input on that.  Thanks.

flex-lm, 2016-01-04:
By flagging any compression or file system related packages as optional; the automatic extraction functionality of said file formats would be lost, this is not a big issue in my opinion.

I would also suggest to flag sasquatch as optional as it's still in beta.

devttys0, 2016-01-04:
I would highly recommend requiring the `python-lzma` module for Python 2.x (this module is standard in Python 3) as it is used to very effectively help invalidate false positive LZMA signatures.

I would also recommend that requiring `pyqtgraph` and `capstone` so that all of binwalk's core functionality is available (entropy graphing and disassembly scans, respectively).

All external extraction utilities can be made optional.

swt2c, 2016-01-04:
Thanks for the feedback.  I'm building binwalk with Python 3 so LZMA is not an issue.

I'd rather not require pyqtgraph as this would essentially pull in a graphical environment (I'm thinking of the case where someone might want to run binwalk on a server or something).  I'll look at requiring capstone.

Thanks.

devttys0, 2016-01-04:
Making either or both `pyqtgraph` and `capstone` optional is fine, just not all options will be available. Honestly, I don't think many people make use of those options anyway, so most users wouldn't notice. :)

closed 2016-01-04 by devttys0