ReFirmLabs/binwalk #178: support extracting filesystem from kyocera printer firmware

Yannik, 2016-01-05:
Hi,

I am trying to extract a kyocera printer firmware. 

Running `binwalk -e` on the firmware file returns the following results:
```
DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
15917         0x3E2D          Zlib compressed data, default compression
13812163      0xD2C1C3        mcrypt 2.5 encrypted data, algorithm: "", keysize: 27369 bytes, mode: "M",
26231768      0x19043D8       LZMA compressed data, properties: 0x63, dictionary size: 16777216 bytes, uncompressed size: 2883884 bytes
26632662      0x19661D6       MySQL ISAM compressed data file Version 1
28384990      0x1B11EDE       YAFFS file system, big endian
```

binwalk extracts the a file `3E2D.zlib` from 0x3E2D to the end of the file and uncompresses it to a file `3E2D`. `strings 3E2D` shows many familiar strings belonging to ELFs and other files (static assets for the webinterface).

`binwalk -e 3E2D` results in > 10GB of .7z files
output:
```
30629505      0x1D35E81       LZMA compressed data, properties: 0x76, dictionary size: 16777216 bytes, uncompressed size: 803209216 bytes
30629825      0x1D35FC1       LZMA compressed data, properties: 0xB6, dictionary size: 16777216 bytes, uncompressed size: 266403840 bytes
30631305      0x1D36589       LZMA compressed data, properties: 0x7E, dictionary size: 16777216 bytes, uncompressed size: 467992576 bytes
....
```
the uncompressed size is unrealistic so these must be false positives.

To exclude these I run `binwalk -e -x 'lzma compressed data' 3E2D`.
This shows a `VxWorks WIND kernel version "2.9"`, various ELFs, Unix Paths and other files, including a 
```
31559244      0x1E18E4C       Linux EXT filesystem, rev 0.0, ext2 filesystem data (mounted or unclean), UUID=00000000-0000-0003-cce2-000234bb34bb
```
entry quite in the beginning. The extracted file `1E18E4C.ext` contains many interesting strings belonging to cgi ELFs and static web assets.

`sudo mount -t ext2 1E18E4C.ext mntdir` reports a bad superblock. I tried to find a valid superblock using `testdisk`, but that didn't succeed either. 

Any ideas on how to properly extract this filesystem?

The firmware zip can be downloaded here: 
https://mega.nz/#!bxg1QLrC!SnEZm4xUHeGmpRVP9CuzYWtXYVHS3PH7FPFv4K7A9d0

devttys0, 2016-01-05:
I don't get the LZMA false positives that you do; your system probably doesn't have the `python-lzma` module installed which binwalk uses to further validate LZMA signatures.

All the file system results are almost certainly false positives. As binwalk suggests, this firmware is VxWorks based, not Linux. Although VxWorks supports some file systems, my experience with it is that there are typically no file systems (your suggestion of FAT16 is possible), at least not as most people would think of them. There are often just some basic data structures with file names, pointers to the file data, and a size field to specify the file size. These are usually used for storing HTML data for the web interface, as the file paths and names that you posted suggest.

I did notice an unusual number of ELF files for a VxWorks image, but they are valid ELFs complete with symbol tables, and looking at the strings and function names they are VxWorks specific. AFAICT, these are VxWorks executables, probably intended to be executed by a user from some command line interface (UART, telnet, ssh, etc).

The main code though is the VxWorks kernel, which appears to be the decompressed `3E2D` file. Both `3E2D` and the main firmware image contain big endian PowerPC code (according to `binwalk -A`), and may or may not contain symbol tables or other helpful information to aid in disassembly  / reverse engineering. These are typically blobs of code and data mashed together, so you'll need to identify where code ends and data starts (`binwalk -A` can help point you in the right direction), as well as the code load address in order to get a readable disassembly.

Yannik, 2016-01-05:
>I don't get the LZMA false positives that you do; your system probably doesn't have the python-lzma module installed which binwalk uses to further validate LZMA signatures.

I just installed python-backports-lzma and was able to `from backports import lzma`. However, binwalk still shows the false positive lzma matches. Using python3 it doesn't. Maybe this is a bug that should be fixed.

As the device supports a huge number of protocols (SNMPv3, Netware, AppleTalk, NetBEUI, Raw Printing, IPP, IPPS, HTTP/HTTPS, FTP, SNMP, POP3, SMTP, LPD, WSD, IPSec, 802.1x), a larger count of ELFs seems reasonable to me. I will maybe investigate into the processor model to find information about the load address later on, the symbol tables being intact is a positive find. Regularly, there is no cli access to users. 
Is it quite certain that everything is baked into the kernel? It would be a bit easier to investigate separated files/modules if it was possible to extract a usable filesystem.

devttys0, 2016-01-05:
>I just installed python-backports-lzma...binwalk still shows the false positive lzma matches. Using python3 it doesn't. Maybe this is a bug that should be fixed.

Binwalk expects the module to be named `lzma`, not `backports.lzma`. Most Linux distros have a `python-lzma` package for Python2; it is a standard module in Python3, which is why you don't get the LZMA false positives in Python3. I've updated the master branch to check for `backports.lzma`, so that should be fixed now.

>a larger count of ELFs seems reasonable to me.

Well, bear in mind when I say "larger" I mean there are 5 or 6 ELF files. Usually with VxWorks I've only seen one ELF file, which was for the decompression stub, so it was not part of the main VxWorks OS.

> the symbol tables being intact is a positive find

The symbol tables for the ELF files are intact, but I didn't immediately see a symbol table for the main OS.

>Regularly, there is no cli access to users.

I did a very brief disassembly of some of the ELF files, and some have command line usage strings and `getopt` style strings in them, so they do take command line options. Could just be for developer debug access via a serial port or something though.

>Is it quite certain that everything is baked into the kernel? It would be a bit easier to investigate separated files/modules if it was possible to extract a usable filesystem.

It would be a lot easier, but unfortunately that's usually how Real Time Operating Systems (like VxWorks) work: the OS is typically one big kernel running in a flat memory space (there is no user space or user applications). With that said, some RTOS's support user space protected code execution, but I'm not sure how that would work in VxWorks.

Yannik, 2016-01-07:
Thanks for your input! I will close this for now.

Yannik, 2016-01-07:
It's backports.lzma on Fedora, probably on other Red Hat systems as well.

closed 2016-01-07 by Yannik