ReFirmLabs/binwalk #180: UnboundLocalError: local variable 'recurse' referenced before assignment

anonymouz, 2016-01-06:
Trying to analyze a unknown partition;
```
$ sudo binwalk -vMe /dev/sda2

Scan Time:     2016-01-06 20:24:38
Target File:   /dev/sda2
MD5 Checksum:  7b1fb330a481f9d0acf1f14fe76a3aa4
Signatures:    345

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------

Signature Exception: local variable 'recurse' referenced before assignment
----------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 566, in main
    retval = self.run()
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/signature.py", line 177, in run
    self.scan_file(fp)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/signature.py", line 160, in scan_file
    self.result(r=r)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 461, in result
    getattr(self, dependency.attribute).callback(r)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/extractor.py", line 172, in callback
    (extraction_directory, dd_file, scan_extracted_files) = self.extract(r.offset, r.description, r.file.path, size, r.name)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/modules/extractor.py", line 511, in extract
    return (output_directory, fname, recurse)
UnboundLocalError: local variable 'recurse' referenced before assignment
----------------------------------------------------------------------------------------------------
$
```
Without looking at the code; Looks like that the 'recurse' variable "escaped" some condition...

devttys0, 2016-01-06:
Yup, latest commit should fix this!

closed 2016-04-13 by devttys0

ahippo, 2018-10-18:
For future reference, it's commit 9882a629c6967e9824aedde94d8107205bc09a42.