ReFirmLabs/binwalk #181: Doesn't detect Linux kernel ARM boot executable zImage (big-endian)

Manawyrm, 2016-01-08:
Hi,

what I have here is a zImage file I manually extracted out of a NAND flash dump from a router. 
running "file kernel.bin" on the file gives me this result: 
```
kernel.bin: Linux kernel ARM boot executable zImage (big-endian)
```

binwalk on the other hand didn't detect the file in the flash dump. 
I'm not sure if this helps, but here's the beginning of the kernel zImage. 

```
$ hexdump kernel.bin | head
0000000 a0e1 0000 a0e1 0000 a0e1 0000 a0e1 0000
0000020 00ea 0200 6f01 1828 0000 0000 1500 484d
0000030 a0e1 0170 a0e1 0280 0fe1 0020 12e3 0300
0000040 001a 0100 a0e3 1700 12ef 5634 0fe1 0020
0000050 82e3 c020 21e1 02f0 11ee 100f 80e3 8000
0000060 01ee 100f 0000 0000 0000 0000 0000 0000
0000070 0000 0000 0000 0000 0000 0000 0000 0000
0000080 a0e1 0f40 04e2 3e43 84e2 0649 00eb 4b00
0000090 8fe2 ec00 90e8 4e1a 90e5 1cd0 40e0 0100
```

If you need the complete file I can send it to you privately, I don't really want to upload the kernel for legal reasons. (although it should be GPLv3, shouldn't it?) 

Thanks,
Tobias

devttys0, 2016-01-09:
Binwalk has signatures for this, but it was probably being incorrectly flagged as a false positive. Should be fixed in the latest commit!

Manawyrm, 2016-01-10:
Thank you very much,
it now works perfectly :+1: 

```
$ binwalk chip1.bin 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             uImage header, header size: 64 bytes, header CRC: 0xA5EE9861, created: 2013-10-18 04:00:25, image size: 9004 bytes, Data Address: 0x0, Entry Point: 0x0, data CRC: 0x295B918C, OS: Linux, CPU: PowerPC, image type: Script file, compression type: none, image name: "Boot Script File"
9068          0x236C          uImage header, header size: 64 bytes, header CRC: 0xFCD4D030, created: 2013-10-18 04:00:25, image size: 10766420 bytes, Data Address: 0x80A00000, Entry Point: 0x80A00000, data CRC: 0x193D4249, OS: Linux, CPU: ARM, image type: Multi-File Image, compression type: none, image name: "Multi Image File"
9144          0x23B8          Linux kernel ARM boot executable zImage (big-endian)
1405952       0x157400        Squashfs filesystem, little endian, version 4.0, compression:xz, size: 9369004 bytes, 1820 inodes, blocksize: 131072 bytes, created: 2013-10-18 04:00:24
```

closed 2016-01-10 by Manawyrm