ReFirmLabs/binwalk #183: deps.sh: wrong pip command?

voltagex, 2016-01-19:
After (trying to) build lzma, the script seems to use the wrong pip invocation for a Debian/python 3 system. 

```
./deps.sh: line 34: pip: command not found
```

```
Manually checking shows the problem
```
voltagex@devbox:~/src/binwalk$ pip
-bash: pip: command not found
voltagex@devbox:~/src/binwalk$ pip3 -V
pip 1.5.6 from /usr/lib/python3/dist-packages (python 3.4)
```

```
voltagex@devbox:~/src/binwalk$ cat /etc/debian_version
8.2
```

Git rev is 441e2d1f527798d8e770ef0c2750239ef8be6028

devttys0, 2016-01-19:
`deps.sh` should have installed `python-pip` via `apt-get` before attempting to install pip packages, so it should exist on your system unless there was an error with the apt-get command. The particular dependency that is referenced on line 34 is for an external extraction utility that runs in python2, which is why pip (instead of pip3) was used.

Were there previous errors or warnings from the deps.sh script, specifically related to apt-get?

voltagex, 2016-01-19:
Re-testing on a brand new Debian 8.2 VM with only git and ssh installed in addition to the base system:
```
voltagex@debian:~/binwalk$ ./deps.sh

WARNING: This script will download and install all required and optional dependencies for binwalk.
         This script has only been tested on, and is only intended for, Debian based systems.
         Some dependencies are downloaded via unsecure (HTTP) protocols.
         This script requires internet access.
         This script requires root privileges.

Continue [y/N]? y
checking for sudo...yes
checking for wget...yes
checking for tar...yes
checking for python...yes
checking for apt-get...yes
checking for python3...no
[sudo] password for voltagex:
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
sudo: pip: command not found
sudo: pip: command not found
Cloning into 'sasquatch'...
remote: Counting objects: 1794, done.
remote: Compressing objects: 100% (3/3), done.
```

voltagex, 2016-01-20:
Fixed in #184

devttys0, 2016-01-20:
Doh! Good catch!

closed 2016-04-13 by devttys0