ReFirmLabs/binwalk #185: PKZIP extraction - Binwalk-2.1.1 fails where Binwalk-2.0.0 succeed

ronan-fontenay, 2016-01-20:
Hello, 

Sum-up : Binwalk-2.1.1 fails where Binwalk-2.0.0 succeed to extract a *.bin firmware.

File :  http://cd.gocam.so/FM/system/CH-sys-48.53.64.67.zip
OS : Linux vpsXXXXXX.ovh.net 3.2.0-4-amd64 #1 SMP Debian 3.2.73-2+deb7u1 x86_64 GNU/Linux

**Binwalk 2.0.0**
Installation : 
```shell
ron@vpsXXXXXX:~/binwalk-2.0.0$ ./deps.sh
ron@vpsXXXXXX:~/binwalk-2.0.0$ ./configure
ron@vpsXXXXXX:~/binwalk-2.0.0$ make
root@vpsXXXXXX:~/binwalk-2.0.0# make install
```

Extraction: 
```shell
ron@vpsXXXXXX:~/firmware$ ls -l
total 1200
-rw-r--r-- 1 ron ron 612699 Jan 20 01:05 CH-sys-48.53.64.67.bin
-rw-r--r-- 1 ron ron 605739 Jan 20 01:05 CH-sys-48.53.64.67.zip

ron@vpsXXXXXX:~/firmware$ binwalk -Mer CH-sys-48.53.64.67.bin

Scan Time:     2016-01-20 01:12:58
Target File:   /home/ron/firmware/CH-sys-48.53.64.67.bin
MD5 Checksum:  58df9214226cfe46760215bfca0c496c
Signatures:    344

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
172           0xAC            Zip archive data, at least v2.0 to extract, compressed size: 8969, uncompressed size: 19091, name: system/system/lib/libsns_gc1004.so
9337          0x2479          End of Zip archive
9499          0x251B          Zip archive data, at least v2.0 to extract, compressed size: 7813, uncompressed size: 16341, name: system/system/lib/libsns_ov9712_plus.so
17518         0x446E          End of Zip archive
17680         0x4510          Zip archive data, at least v2.0 to extract, compressed size: 90121, uncompressed size: 353248, name: system/system/lib/libOnvif.so
107987        0x1A5D3         End of Zip archive
108149        0x1A675         Zip archive data, at least v2.0 to extract, compressed size: 43603, uncompressed size: 84480, name: system/system/lib/libvoice_arm.so
151946        0x2518A         End of Zip archive
152108        0x2522C         Zip archive data, at least v2.0 to extract, compressed size: 130, uncompressed size: 227, name: system/init/ipcam.sh
152406        0x25356         End of Zip archive
152568        0x253F8         Zip archive data, at least v2.0 to extract, compressed size: 402383, uncompressed size: 886168, name: system/system/bin/encoder
555129        0x87879         End of Zip archive
555291        0x8791B         Zip archive data, at least v2.0 to extract, compressed size: 35394, uncompressed size: 74200, name: system/system/bin/wifidaemon
590869        0x90415         End of Zip archive
591031        0x904B7         Zip archive data, at least v2.0 to extract, compressed size: 1852, uncompressed size: 9692, name: system/system/bin/grade.sh
593063        0x90CA7         End of Zip archive
593225        0x90D49         Zip archive data, at least v2.0 to extract, compressed size: 8704, uncompressed size: 20212, name: system/system/bin/updata
602105        0x92FF9         End of Zip archive
602267        0x9309B         Zip archive data, at least v2.0 to extract, compressed size: 1874, uncompressed size: 4522, name: system/system/bin/gpio_aplink.ko
604333        0x938AD         End of Zip archive
604495        0x9394F         Zip archive data, at least v2.0 to extract, compressed size: 7241, uncompressed size: 16802, name: system/system/bin/motogpio.ko
611922        0x95652         End of Zip archive
612084        0x956F4         Zip archive data, at least v1.0 to extract, compressed size: 8, uncompressed size: 8, name: system/system/bin/fwversion.bin
612282        0x957BA         End of Zip archive
612444        0x9585C         Zip archive data, at least v1.0 to extract, compressed size: 9, uncompressed size: 9, name: system/system/bin/sysversion.txt
612645        0x95925         End of Zip archive

Scan Time:     2016-01-20 01:12:58
Target File:   /home/ron/firmware/_CH-sys-48.53.64.67.bin.extracted/system/system/bin/sysversion.txt
MD5 Checksum:  3e98d83fbced8eb62c79542f5df5a14f
Signatures:    344

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------

ron@vpsXXXXXX:~/firmware$ tree _CH-sys-48.53.64.67.bin.extracted/
_CH-sys-48.53.64.67.bin.extracted/
└── system
    └── system
        └── bin
            └── sysversion.txt

3 directories, 1 file
```
**Binwalk 2.1.1**

Installation : 
```shell
ron@vpsXXXXXX:/home/ron/binwalk-2.1.1$ ./deps.sh
root@vpsXXXXXX:/home/ron/binwalk-2.1.1# python setup.py install
```

Extraction: 
```shell
ron@vpsXXXXXX:~/firmware$ ls -l
total 1200
-rw-r--r-- 1 ron ron 612699 Jan 20 01:05 CH-sys-48.53.64.67.bin
-rw-r--r-- 1 ron ron 605739 Jan 20 01:05 CH-sys-48.53.64.67.zip
ron@vpsXXXXXX:~/firmware$ binwalk -Mer CH-sys-48.53.64.67.bin

Scan Time:     2016-01-20 01:12:58
Target File:   /home/ron/firmware/CH-sys-48.53.64.67.bin
MD5 Checksum:  58df9214226cfe46760215bfca0c496c
Signatures:    344

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
172           0xAC            Zip archive data, at least v2.0 to extract, compressed size: 8969, uncompressed size: 19091, name: system/system/lib/libsns_gc1004.so
9337          0x2479          End of Zip archive
9499          0x251B          Zip archive data, at least v2.0 to extract, compressed size: 7813, uncompressed size: 16341, name: system/system/lib/libsns_ov9712_plus.so
17518         0x446E          End of Zip archive
17680         0x4510          Zip archive data, at least v2.0 to extract, compressed size: 90121, uncompressed size: 353248, name: system/system/lib/libOnvif.so
107987        0x1A5D3         End of Zip archive
108149        0x1A675         Zip archive data, at least v2.0 to extract, compressed size: 43603, uncompressed size: 84480, name: system/system/lib/libvoice_arm.so
151946        0x2518A         End of Zip archive
152108        0x2522C         Zip archive data, at least v2.0 to extract, compressed size: 130, uncompressed size: 227, name: system/init/ipcam.sh
152406        0x25356         End of Zip archive
152568        0x253F8         Zip archive data, at least v2.0 to extract, compressed size: 402383, uncompressed size: 886168, name: system/system/bin/encoder
555129        0x87879         End of Zip archive
555291        0x8791B         Zip archive data, at least v2.0 to extract, compressed size: 35394, uncompressed size: 74200, name: system/system/bin/wifidaemon
590869        0x90415         End of Zip archive
591031        0x904B7         Zip archive data, at least v2.0 to extract, compressed size: 1852, uncompressed size: 9692, name: system/system/bin/grade.sh
593063        0x90CA7         End of Zip archive
593225        0x90D49         Zip archive data, at least v2.0 to extract, compressed size: 8704, uncompressed size: 20212, name: system/system/bin/updata
602105        0x92FF9         End of Zip archive
602267        0x9309B         Zip archive data, at least v2.0 to extract, compressed size: 1874, uncompressed size: 4522, name: system/system/bin/gpio_aplink.ko
604333        0x938AD         End of Zip archive
604495        0x9394F         Zip archive data, at least v2.0 to extract, compressed size: 7241, uncompressed size: 16802, name: system/system/bin/motogpio.ko
611922        0x95652         End of Zip archive
612084        0x956F4         Zip archive data, at least v1.0 to extract, compressed size: 8, uncompressed size: 8, name: system/system/bin/fwversion.bin
612282        0x957BA         End of Zip archive
612444        0x9585C         Zip archive data, at least v1.0 to extract, compressed size: 9, uncompressed size: 9, name: system/system/bin/sysversion.txt
612645        0x95925         End of Zip archive

Scan Time:     2016-01-20 01:12:58
Target File:   /home/ron/firmware/_CH-sys-48.53.64.67.bin.extracted/system/system/bin/sysversion.txt
MD5 Checksum:  3e98d83fbced8eb62c79542f5df5a14f
Signatures:    344

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------

ron@vpsXXXXXX:~/firmware$ tree _CH-sys-48.53.64.67.bin.extracted/
_CH-sys-48.53.64.67.bin.extracted/
└── system
    └── system
        └── bin
            └── sysversion.txt

3 directories, 1 file
```

devttys0, 2016-01-20:
This should now be fixed in the latest master branch. You will need the `jar` utility (part of the `openjdk-7-jdk` package) if you do not already have it installed.

ronan-fontenay, 2016-01-20:
The 2.0.0 did the job so it's fine for me. I just mentioned it in case you were not aware.
Great to know that it'll be fixed in the next stable version.

closed 2016-01-20 by ronan-fontenay