ReFirmLabs/binwalk #189: Errno 2 - No such file or directory

xujianningkp, 2016-02-03:
Hello all,

Currently running test with Binwalk v2.1.2b, but keep getting the same error when I run "binwalk --signature XXXX.bin" Can any one explain the follow error? How can I address it? Thank you.

Traceback (most recent call last):
  File "/usr/local/bin/binwalk", line 57, in <module>
    main()
  File "/usr/local/bin/binwalk", line 36, in main
    elif not modules.execute():
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 729, in execute
    obj = self.run(module)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 746, in run
    obj = self.load(module, kwargs)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 774, in load
    argv.update(self.dependencies(module, argv['enabled']))
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 800, in dependencies
    depobj = self.run(dependency.module, dependency=True, kwargs=dependency.kwargs)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 746, in run
    obj = self.load(module, kwargs)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 775, in load
    return module(self, **argv)
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/module.py", line 238, in __init__
    self.plugins.load_plugins()
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/plugin.py", line 189, in load_plugins
    plugins = self.list_plugins()
  File "/usr/local/lib/python2.7/dist-packages/binwalk/core/plugin.py", line 164, in list_plugins
    for file_name in os.listdir(plugins[key]['path']):
OSError: [Errno 2] No such file or directory: '/home/XXXX/.binwalk/plugins/'

devttys0, 2016-02-04:
Binwalk expects there to be a `$HOME/.binwalk` directory, where users can put their own custom plugins and signatures. Binwalk will attempt to create these directories if they don't already exist. It seems that binwalk was unable to create them, so they don't exist when binwalk looks for them.

Are there some permissions settings for your current user account that might prevent binwalk from creating the `$HOME/.binwalk` directories?

xujianningkp, 2016-02-04:
Thank you for the information!

I have to run sudo binwalk in order to get it to work :)

closed 2016-04-13 by devttys0