ReFirmLabs/binwalk #190: Binwalk applied to an initrd/initramfs image

CMCDragonkai, 2016-02-13:
I had a Linux based initramfs/initrd file. Applying binwalk to it, showed me 2 gzip files that I wanted. The first gzip file "A1E00" was a gzipped cpio archive that I injected into the initrd previously. And the second gzip file "2E1834" was also a gzipped cpio archive that was the actual root directory of the initramfs.

After extracting these 2 using `--extract`, I found out that both of these files had a similar size to them. In fact `ll -h` showed that the "A1E00" is 338MiB, and that "2E1834" is 334MiB. 

This was very interesting, because I know in fact that my first gzip file contained a 4MiB image when I added it in. Running `binwalk` on these 2 cpio archives showed that the 2 files were the exact same except "A1E00" had 2 extra pieces on it, it was the 4MiB image that I created and a "TRAILER!!!" file, which according to the initramfs format indicates the end of a CPIO archive.

So my question would be, why did binwalk extract 2 gzipped cpio archives, but didn't (I'm guessing) properly partition them? That is "A1E00" essentially contains "2E1834"? If I try to extract the CPIO "A1E00", it only extracts until the "TRAILER!!!", essentially making the rest of the file useless.

CMCDragonkai, 2016-02-13:
Later I discovered that if I manually use `dd` to cut out the gzipped files. It works properly and I don't get 2 massive files, just 1 ~4MiB file and a ~334MiB file. So it does seem like binwalk is correctly carving up the 2 gzipped files.

devttys0, 2016-02-16:
It sounds like you have two gzip files back-to-back, with no data in between. When `gzip` sees a valid gzip header right at the end of some other gzip data, it treats them as one file, which is why your first decompressed file contains both your 4MB file and your 334MB file.

Binwalk doesn't know where the first gzip file should end, so it just extracts *everything* from the start of the gzip header to the EOF. It could stop at the entry of the second gzip file, but that's problematic if the second gzip header was actually a false positive, something which binwalk can't know for sure.

In practice, I haven't seen this cause any major issues, and in your specific case it sounds like binwalk was still able to extract both CPIO archives.

CMCDragonkai, 2016-02-17:
I suppose the 2nd gzip could be a false positive, but I'd expect the extraction to follow the table of contents shown (and allow the user to change if it is indeed a false positive). Perhaps a better documentation on this would be helpful then.

closed 2018-04-12 by CoffeeExpress