ReFirmLabs/binwalk #193: can't identify hlktech firmware image signatures in upgrade file

jameshilliard, 2016-03-25:
It seems this device is using some sort of custom packing method for [stock upgrade images](https://github.com/jameshilliard/HLK-SW16-FW/raw/master/HLK_WR02_v1.95_20140611_2.img), it has an upload.cgi binary that is similar to [this](https://github.com/socoola/yhrouter/blob/master/user/goahead/cgi-src/upload.cgi.c) from what I can tell. I was able to dump the mtd partitions from a running device and [mtd5](https://github.com/jameshilliard/HLK-SW16-FW/raw/master/mtd5.bin) [extracts](https://github.com/jameshilliard/hlk-wr02-squashfs-root) just fine as far as I can tell as squashfs-root.

jameshilliard, 2016-03-25:
Looks like it's DES encrypted, I was able to decrypt it with [this](https://raw.githubusercontent.com/openwrt-mirror/openwrt/f0c979509e3c579811918ec7fb201be8fcc246ed/tools/firmware-utils/src/mkhilinkfw.c) and then extract it with binwalk.

devttys0, 2016-03-31:
Nice! The latest version of binwalk in the master branch can now identify and extract these encrypted Hilink firmware images (requires the `pycrypto` Python library).

jameshilliard, 2016-03-31:
This extracts the stock image in my origional post but doesn't appear to extract [other images](https://github.com/jameshilliard/HLK-SW16-FW/blob/master/RM04_v1.78%20firmware.img) with the same encryption.

devttys0, 2016-04-01:
Those encrypted bytes seem to correspond to the "HLK-" portion of the description field in the uImage header, maybe? If so, provided the firmware uses that same naming convention, that should be a reliable signature. I've updated the master branch accordingly.

jameshilliard, 2016-04-01:
Yeah, that's what it looks like to me at least, wonder if it would be better to target the bytes after since they appear to be all 0's when decrypted in case the HLK field changes. It seems quite strange that a no-name Chinese company would actually encrypt their firmware images like this, have you seen any other companies do that before?

closed 2016-04-23 by jameshilliard