ReFirmLabs/binwalk #195: Commit 5fdad3c5ffe63efc1f0b213a56dd962477f7190f breaks extraction with firmware located outside the CWD

ddcc, 2016-04-01:
Commit 5fdad3c5ffe63efc1f0b213a56dd962477f7190f refactored some of the base directory computation code in `extractor.py`:419 within `build_output_directory`. However, calling `basedir.split(self.directory)` assumes that the current working directory is located in a subtree of the base directory, which isn't necessarily true. Doing string matching on paths seems liable to break, especially when symbolic links are involved. This can be reproduced by executing: `cd ~; touch foo; gzip foo; cd /tmp/; binwalk -e ~/foo.gz`.

Found by @misterch0c, originally reported at https://github.com/firmadyne/firmadyne/issues/4.

devttys0, 2016-04-01:
Thanks, commit 85c522b97f7b3a365f0e10432c9e37329a882c18 should fix this.

misterch0c, 2016-04-02:
I confirm it works now. Thank you guys, that was fast =D

closed 2016-04-04 by ddcc