ReFirmLabs/binwalk #196: Lzma uncompressed size : -1

YaseenTwati, 2016-04-07:
Im trying to extract a firmware bin file but Binwalk returns -1 as the size of the uncompressed lzma archive, and outputs an empty file when trying to extract it 
what could be happening right here ? i dont think he firmware is corrupted or somthing

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
14944         0x3A60          LZMA compressed data, properties: 0x6D, dictionary size: 8388608 bytes, uncompressed size: -1 bytes
854872        0xD0B58         CramFS filesystem, little endian, size: 4202496 version 2 sorted_dirs CRC 0xAFF93F00, edition 16777728, 268466059 blocks, 1392509248 files

kamil-HFC, 2016-04-08:
Can you upload this firmware?

YaseenTwati, 2016-04-08:
@kamil-HFC .. ok .. its from the companies website so theres no way its corrupted 
http://www.olo.com.pe/OTA/IMW-C910W/IMW-C91HW_V2324_R4932A.bin

kamil-HFC, 2016-04-08:
Thanks. I extracted it with binwalk -e without problems. Maybe try binwalk re-installation? 
extracted file has 3080380 bytes and inside ot it:

`$ binwalk 3A60

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
76576         0x12B20         gzip compressed data, maximum compression, from Unix, last modified: 2014-04-07 00:32:56
1855440       0x1C4FD0        Linux kernel version "2.6.23-uc1 (jhpark@build-server) (gcc version 3.4.4) #1 Mon Apr 7 09:33:21 KST 2014"
1886728       0x1CCA08        CRC32 polynomial table, little endian
2392082       0x248012        LZMA compressed data, properties: 0x6D, dictionary size: 16777216 bytes, uncompressed size: 224 bytes
2544202       0x26D24A        LZMA compressed data, properties: 0x6D, dictionary size: 16777216 bytes, uncompressed size: 236 bytes
2620502       0x27FC56        LZMA compressed data, properties: 0x6D, dictionary size: 16777216 bytes, uncompressed size: 224 bytes
2772622       0x2A4E8E        LZMA compressed data, properties: 0x6D, dictionary size: 16777216 bytes, uncompressed size: 236 bytes
2848922       0x2B789A        LZMA compressed data, properties: 0x6D, dictionary size: 16777216 bytes, uncompressed size: 224 bytes
3001042       0x2DCAD2        LZMA compressed data, properties: 0x6D, dictionary size: 16777216 bytes, uncompressed size: 236 bytes`

I uploaded file for you. http://www30.zippyshare.com/v/qrPulCgF/file.html

kamil-HFC, 2016-04-08:
And I think, binwalk recognized LZMA as -1 uncompressed size, because an archive is streamed, and 8 bytes used for uncompressed-size describing are filled by FF:
`00000000  5d 00 00 80 00 ff ff ff  ff ff ff ff ff 00 36 80  |].............6.|`

YaseenTwati, 2016-04-08:
Oh WoW .. thanks ALOT , i had problems with this for long time , its actually pretty weird .. as my friend got the exact same results ...
and the firmware is very weird as every compressed file is just filled with other compressed files .. :|
what version of binwalk are you using ?

kamil-HFC, 2016-04-08:
Your welcome. 
It's just a coincidence, but today (before i saw your issue) i removed old 2.0 version and installed newest v2.1.2b by  setup.py install and then ./deps.sh. As Graig suggest to update, because 2.0.0 is "pretty old" :) https://github.com/devttys0/binwalk/issues/188

closed 2016-04-13 by devttys0

okjokj2, 2021-02-05:
i need the file IMW-C91HW_V2324_R4932A.bin please uploaded it .. ill extract it for sure