ReFirmLabs/binwalk #198: Smart signatures format changed

m0sia, 2016-04-13:
I noticed that smart signatures format has been changed from {file-size}/{adjust-offset} to {size}/{adjust}.

I am trying to port some of my old signatures and {size} doesn't seems to work the same way I used it before.

Example signature file:
```
0    string    file     file header
>4	belong x {size:%d}size: %d
>8	belong x {adjust:%d}offset: %x
```

Extract example:
```
$binwalk -m custom --dd "file header:bin:" firmware.bin
963           0x3C3          file header size: 43776 offset: 382
44791         0xAEF7          file header size: 790016 offset: ae82
834859        0xCBD2B         file header size: 93696 offset: cbc82
```

I expect extracted 3C3.bin file to be 43775 bytes, but it is 1048576 instead.

m0sia, 2016-04-13:
It seems that it is not about smart signatures, but about "-dd" option carving files with "1048576"(1024*1024) block size.

If {size} in signature result is bigger then (1024*1024), then resulting file will be 2097152(1024*1024*2). I quickly searched the code, but I can't find the place where I can define carving block size.

devttys0, 2016-04-13:
This was a bug in the `binwalk.modules.extractor._dd` method, which rounded up to the nearest block size. Latest commit should fix this.

m0sia, 2016-04-13:
Thanks for fixing that

closed 2016-04-13 by m0sia