ReFirmLabs/binwalk #200: Options not available under current master?

surfrock66, 2016-04-17:
I just ran this right now:
`wget https://github.com/devttys0/binwalk/archive/master.zip; unzip master.zip; (cd binwalk-master && sudo python setup.py uninstall && sudo python setup.py install)`

It installed and runs successfully, showing the following:

`Binwalk v2.1.2b`

It appears several options, such as --deflate or --lzma, are not available in this version; they result in the following:

`surfrock66@sr66-darter:~/Downloads$ binwalk --deflate chipdump1

General Error: Cannot open file : [Errno 2] No such file or directory: '--deflate'`

These options also do not appear in the help text.  Is there something I must do to enable those?

devttys0, 2016-04-18:
Some binwalk options are disabled if their required dependencies are not installed on the local system. Adding some output indicating which options are disabled is on my todo list, but I haven't gotten around to it yet.

Binwalk's raw decompression module (which handles the `--deflate` and `--lzma` options) requires the Python lzma module, which is not installed by default with Python 2.x. Python3 does include the lzma module by default, or you can install the lzma module for Python2 (e.g., `sudo apt-get install python-lzma`).

surfrock66, 2016-04-19:
Hmm; here was my full install command set:

`sudo apt-get install python-lzma python-crypto python3-crypto libqt4-opengl python-opengl python-qt4 python-qt4-gl python-numpy python-scipy python-pip python3-opengl python3-pyqt4 python3-pyqt4.qtopengl python3-numpy python3-scipy python3-pip mtd-utils gzip bzip2 tar arj lhasa p7zip p7zip-full cabextract cramfsswap squashfs-tools zlib1g-dev liblzma-dev liblzo2-dev liblzo2-dev python-lzo`
`sudo pip install pyqtgraph capstone cstruct`
`sudo pip3 install pyqtgraph capstone cstruct`
`cd ~/.sources`
`git clone https://github.com/devttys0/sasquatch`
`(cd sasquatch && ./build.sh)`
`cd ~/.sources`
`git clone https://github.com/sviehb/jefferson`
`(cd jefferson && sudo python setup.py install)`
`cd ~/.sources`
`git clone https://github.com/jrspruitt/ubi_reader`
`(cd ubi_reader && sudo python setup.py install)`
`cd ~/.sources`
`git clone https://github.com/devttys0/yaffshiv`
`(cd yaffshiv && sudo python setup.py install)`
`cd ~/.sources`
`wget -O - http://my.smithmicro.com/downloads/files/stuffit520.611linux-i386.tar.gz | tar -zxv`
`sudo cp bin/unstuff /usr/local/bin/`
`cd ~/.sources`
`git clone https://github.com/devttys0/binwalk`
`sudo python setup.py install`

I'm still getting nothing on -X, -Y, and -Z.  Thoughts?

devttys0, 2016-04-22:
If Python can `import lzma`, then the `-X` and `-Z` options should be enabled. The fact that you aren't getting `-Y` either, even though you installed capstone, which `-Y` relies on, suggests to me that this may be a system-specific issue. Possibly an problem with your Python paths?

As a test, install binwalk under Python3; there, you should at least have the `-X` and `-Z` options, since the lzma module is standard in Python3.

closed 2018-04-12 by CoffeeExpress