ReFirmLabs/binwalk #203: Recursive extraction fails when file type is misidentified

RenaKunisaki, 2016-04-30:
Running `binwalk -eM` on [this file](https://www.dropbox.com/s/nqu0w9e64p7tz9a/bt2700hgv_reinstall_enh_2sp.zip?dl=0) (after unzipping) extracts a lot of `.crt` files, which Binwalk believes to be certificates (despite some are several megabytes), but it does not go on to extract those. They have to be manually extracted by invoking `binwalk` again.

devttys0, 2016-05-03:
Binwalk doesn't know how big the certificate files are, so it just copies out everything from the beginning of the cert to the end of the file. During recursive extraction this can cause problems, since binwalk would extract a bunch of data from the original file, then end up re-scanning it, extracting it again, re-scanning it, extracting it again, etc; hence binwalk refused to extract them when running recursively. In fact, the latest version of binwalk doesn't extract certificate files under any circumstances, since it's easy enough to `dd` them out if you really want them: https://github.com/devttys0/binwalk/blob/master/src/binwalk/config/extract.conf#L89

With that said, I don't think these file types are being mis-identified. At least the first few certificate results that binwalk reported look like legitimate certificate files.

RenaKunisaki, 2016-05-09:
So all the extra files extracted from those .crt files are just duplicates of what's in the original file?

I wonder if it's possible to determine their size by looking at nearby words that might be a size, extracting that much, and checking if the result is a valid certificate (or other type of file) without any extra junk at the end.

closed 2018-04-12 by CoffeeExpress