ReFirmLabs/binwalk #204: Deps

rfminelli, 2016-05-01:
Please, upgrade openjdk-8-jdk to Deps.sh

devttys0, 2016-05-07:
Any specific reason why? While openjdk-7 is old, openjdk-8 is not in the package repos for some older, but still supported, distros (e.g., Ubuntu LTS releases). The deps.sh install script installs openjdk-7 since it is more commonly available.

rfminelli, 2016-05-08:
One option would be an IF for "APT_CANDIDATES" (Debian based) if the repository openjdk-8-jdk or openjdk-9-jdk installation would be done with the most current case I could not be done with openjdk-7-jdk .
I believe Deps.sh would be more intelligent.

devttys0, 2016-05-09:
I agree, deps.sh could be made smarter to handle this. But what is the advantage of version 8/9 over 7?

rfminelli, 2016-05-09:
Initially my idea was only to facilitate those who do not dominate linux console.
But really you talves be right, because if not dominate, can not use the tool.
But I thank the attention.

closed 2016-05-09 by rfminelli