ReFirmLabs/binwalk #205: Package

rfminelli, 2016-05-14:
How can we package the firmware after extracted.
Sorry if the question is the wrong place.

Lekensteyn, 2016-05-20:
I don't think this the goal of binwalk. Using binwalk you can analyze files and execute arbitrary commands based on signature matches. This is a one-way operation, packing it is not possible (and if it were, then it would be possible only in a limited number of cases).

devttys0, 2016-05-21:
As Lekensteyn said, binwalk doesn't do re-construction of firmware after it has been extracted. Doing so is typically a manual process, although there are some tools that can automate this for limited numbers of firmwares (see: https://bitsum.com/firmware_mod_kit.htm).

closed 2016-05-21 by devttys0