ReFirmLabs/binwalk #207: Calling more and more time as call times increases

bingqiangzhang, 2016-05-21:
Hello, I have found that binwalk in a script will call more and more time when we call binwalk again and again.This is my test script.

	cat binwalk_test.py 
	#coding=utf8
	import time
	import sys
	import binwalk
	try:
	    loop_times = int(sys.argv[1])
	except:
	    loop_times = 5
	
	files = ['/etc/hosts'] * loop_times
	
	t0 = time.time()
	for _file in files:
	    t1 = time.time()
	    binwalk.scan(_file, '--signature','-q')
	    t2 = time.time()
	    print _file, t2 - t1
	tx = time.time()
	print tx - t0

And this is the result of my test which calls binwalk for  8 times .

	python binwalk_test.py 8
	/etc/hosts 0.42361998558
	/etc/hosts 0.573173046112
	/etc/hosts 0.740453004837
	/etc/hosts 0.888923883438
	/etc/hosts 1.16715812683
	/etc/hosts 1.34311795235
	/etc/hosts 1.56189513206
	/etc/hosts 1.83435893059
	8.53298807144

As we can see, time used more and more.I know that binwalk can scan many file at a time ,but this is another question.Please show me if you can.Thank you very much!

devttys0, 2016-05-21:
Here's what I get running your script on my Ubuntu 14.04 VM in Python2.7:

```
$ python binwalk_test.py 8
/etc/hosts 0.177801847458
/etc/hosts 0.209730863571
/etc/hosts 0.280726909637
/etc/hosts 0.35257101059
/etc/hosts 0.462613105774
/etc/hosts 0.507649898529
/etc/hosts 0.548542976379
/etc/hosts 0.793392181396
3.33378410339
```

As you can see, I get similar results in terms of a speed decrease over time, but the overall scan time is more than twice as fast as yours. My VM is not the pinnacle of processing power (single core 2.2GHz i7 CPU, 2GB of RAM), so I doubt that the overall speed difference is due to CPU speed. More likely, it is due to my having a solid state hard drive; noticeable speed increases can be seen when switching from a traditional HDD to an SSD.

Now, the reason I mention this is that I suspect (but have not confirmed) that the speed decrease is due to the fact that each time you run `binwalk.scan`, the binwalk module will re-load everything from disk. This includes all the magic files, plugins, etc. Many of these files are small in size and result in lots of small hard-drive accesses, which are generally very inefficient, and execution time from Linux user space is non-deterministic.

In contrast, here is a script that repeatedly scans the `/etc/hosts` file with a single call to `binwalk.scan`:

```
import time
import sys
import binwalk
try:
    loop_times = int(sys.argv[1])
except:
    loop_times = 5

files = ['/etc/hosts'] * loop_times

t0 = time.time()
binwalk.scan(*files, signature=True, quiet=True)
tx = time.time()
print (tx - t0)
```

Scanning the same `/etc/hosts` file the same number of times is now much faster:

```
$ python bwalk_test.py 8
0.194328784943
$ python bwalk_test.py 8
0.187590122223
$ python bwalk_test.py 8
0.191529989243
$ python bwalk_test.py 8
0.184499979019
$ python bwalk_test.py 8
0.185815811157
$ python bwalk_test.py 8
0.179244995117
```

Since the `/etc/hosts` file is being accessed and scanned the same number of times, the main difference in the two scripts is that the binwalk initialization is only done once, which would seem to support my hypothesis. I'll need to do some more detailed benchmark tests to verify one way or the other though.

bingqiangzhang, 2016-05-22:
Thanks for your reply.
> the speed decrease is due to the fact that each time you run binwalk.scan, the binwalk module will re-load everything from disk. 

I also suspect this. And I run another test which use `subprocess.call` to call binwalk by system command with the script below:

	cat binwalk_test.py 
	#coding=utf8
	import time
	import sys
	from subprocess import call
	import binwalk
	
	try:
	    loop_times = int(sys.argv[1])
	except:
	    loop_times = 5
	
	files = ['/etc/hosts'] * loop_times
	
	time_api = []
	time_sys = []
	for _file in files:
	    t1 = time.time()
	    binwalk.scan(_file, '--signature','-q')
	    t2 = time.time()
	    td1 = t2 - t1
	    print('API Call: %s'%td1)
	    call(['binwalk', _file, '-q'])
	    t3 = time.time()
	    td2 = t3 - t2
	    print('SYS Call: %s\n'%td2)
	    time_api.append(td1)
	    time_sys.append(td2)
	print('API Total: %s'%sum(time_api))
	print('SYS Total: %s'%sum(time_sys))

Here is the result 

	python binwalk_test.py 8
	API Call: 0.421117067337
	SYS Call: 0.619277954102
	
	API Call: 0.528059959412
	SYS Call: 0.606225967407
	
	API Call: 0.774738073349
	SYS Call: 0.662249088287
	
	API Call: 0.977311134338
	SYS Call: 0.563318967819
	
	API Call: 1.18678092957
	SYS Call: 0.556361913681
	
	API Call: 1.51776599884
	SYS Call: 0.65579199791
	
	API Call: 1.72820901871
	SYS Call: 0.62273311615
	
	API Call: 2.03798413277
	SYS Call: 0.668081998825
	
	API Total: 9.17196631432
	SYS Total: 4.95404100418

As you see, `subprocess.call` does the same thing, but its speed is not decreased as `binwalk.scan`. It is so amazing.

PR #218 (2016-07-19): Update kwarg processing to ensure modules don't receive a mutable reference to defaults
Fixes #207 , also the Signature module was "leaking" memory.

The `Signature` module `magic_files` attribute is initialized via kwarg.  When no `magic_files` argument is specified (the default mode), it receives a reference to the default empty list. During Signature's init() method, this list is modified, and because it is modifying the class variable default, it continues to grow every time the Signature module executes. This causes a long-running binwalk process to appear to leak memory and the duplicate [... cut]
commit 11b544e1 (2016-07-19): Update kwarg processing to ensure modules don't receive a mutable reference to a class variable default. Fixes #207, also the Signature Modulemodule was leaking memory.
closed 2016-07-19 by devttys0