ReFirmLabs/binwalk #208: [documentation] How to set the size for extraction?

Lekensteyn, 2016-05-21:
The [Signature File Format][1] wiki lists some tags that seem usable for restricting the size of a dumped file, but it does not seem to work in my case. The full file is extracted, so I probably misunderstood how size (and jump) work(s).

Are tags like `offset-adjust` still valid? Commit 9e0a0c1fd7 removed the last user. It would be nice if the documentation is updated (bonus points for an example to set the output size :smiley:).

If you are interested, this is the magic I am using with binwalk 2.1.1.
```
# Match DSDT/SSDT, presumably smaller than 16 MiB (avoids false positives)
0       regex   [SD]SDT ACPI {name:%.4s}%.4s
>4      ulelong x       Length: 0x%08X
>7      ubyte   !0      {invalid}
>8      ubyte   x       Revision: 0x%02X
>9      ubyte   x       Checksum: 0x%02X
>10     string  x       OemId: "%-6.6s"
>16     string  x       OemTableId: "%-8.8s"
>24     ulelong x       OemRevision: 0x%04X
>28     string  x       CreatorId: "%-4.4s"
>32     ulelong x       CreatorRevision: 0x%X
# this is my attempt, first tried {size:%d} in `>4` above and below, that did not work either
>4      ulelong x       {jump:%u}
>4      ulelong x       {size:%u}
```

 [1]: https://github.com/devttys0/binwalk/wiki/Signature-File-Format#tagging-metadata

Lekensteyn, 2016-05-21:
The special tags disappeared when libmagic was replaced in 9557e51.

As for how binwalk currently (0efe0e1) parses the format string (paths are relative to src/binwalk):
- Class `SignatureLine` in core/magic.py parses the line. The description part (fourth field) is optional, but if it exists it will extract and remove all tags, patterns like `{...}` (the resulting string without `{..}` is called the *title* of the signature).
  - If a tag has the form `{name:value}`, then `name` will be used as key pointing to the string `value`.
  - If a tag has the form `{name}`, then `name` will point to the boolean `True` (used for `{invalid}` for example).
- `Magic._analyze` will evaluate such a signature against the data, returning a dictionary of tags. 
  - Initially this wil lcontain the keys **id** (signature.id), **offset** (number), **invalid** (False) and **once** (False).
  - Additional tags from the signature (as extracted by SignatureLine) will also be applied as follows:
    - Strings will be interpreted as format strings and interpolated with a value matching the data type (Unicode strings for `string`, numbers for (u)byte, (u)(be|le)long, etc.). Zero or more percent-specifiers are supported. You could for example use it to format as decimal and hexadecimal: the signature description <code>{**my_tag**:dec=%d hex=%X}</code> could be expanded to `dec=10 hex=A` (which is then stored as tag name **my_tag**).
    - If a string can be converted to a number, it will be done using `int(tag_value, 0)`. So string 10 becomes number 10 and string 0x10 becomes number 16 (hexadecimal conversion).
  - Finally all expanded strings are joined and stored using the tag name **description**.
  - (Also interesting is that empty descriptions and descriptions containing non-printable characters are considered **invalid** (True).)
- `Magic.scan` uses above `Magic._analyze` to build the initial tags *dictionary*. (It also skips the signature match if the tag **once** is True and if the *title* of the signature (see `SignatureLine` above) was seen before).
  - Finally the tags dictionary is converted to a `SignatureResult`, a subclass of `Result` (see `core/module.py`) which is described as "Generic class for storing and accessing scan results".
  - All accumulated `SignatureResults` are stored in a sorted list and returned by `Magic.scan(data)`.
- `Magic.match(data)` is like `Magic.scan(data)`, but it only matches prefixes in the data (`Magic.scan` accepts results starting from any offset in the data).

So, the acceptable tags are **jump**, **adjust**, etc. as can be seen below:
```py
class SignatureResult(binwalk.core.module.Result):
    '''
    Container class for signature results.
    '''
    def __init__(self, **kwargs):
        # These are set by signature keyword tags.
        # Keyword tags can also set any other object attributes,
        # including those in binwalk.core.module.Result.
        self.jump = 0
        self.many = False
        self.adjust = 0
        self.strlen = 0
        self.string = False
        self.invalid = False
        self.once = False
        self.overlap = False
```
and also (from `Result.__init__` in core/module.py, also documented in API.md in the root dir):
```
        @offset      - The file offset of the result.
        @size        - Size of the result, if known.
        @description - The result description, as displayed to the user.
        @module      - Name of the module that generated the result.
        @file        - The file object of the scanned file.
        @valid       - Set to True if the result if value, False if invalid.
        @display     - Set to True to display the result to the user, False to hide it.
        @extract     - Set to True to flag this result for extraction.
        @plot        - Set to Flase to exclude this result from entropy plots.
        @name        - Name of the result found (None if not applicable or unknown).
```

The `offset` and `adjust` tags influence the scanning process in `modules/signature.py` (see `r.adjust` and other properties on `r`).

Finally for extraction, we have this call chain in `Module.result(r)` (`core/module.py`) -> `Extractor.callback(r)` (modules/extractor.py). Here we can find that a tag name such as **size** (`r.size`) *should* in theory work.

Let's see...

devttys0, 2016-11-22:
Finally got around to updating this in the Wiki. Added `jump` and `size` examples too!

closed 2016-11-22 by devttys0