ReFirmLabs/binwalk #210: ImportError: No module named core.module

shadyproject, 2016-05-23:
Trying to use binwalk in a virtualenv, and I get the following error trying to import binwalk 

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/username/src/sahara/venv/lib/python2.7/site-packages/binwalk/__init__.py", line 3, in <module>
    from binwalk.core.module import Modules, ModuleException
ImportError: No module named core.module
```

This occurs both when the venv is activated via ```source venv/bin/activate``` or when the venv python is called directly via ```venv/bin/python```

Thoughts?

devttys0, 2016-06-03:
Does the file `/Users/username/src/sahara/venv/lib/python2.7/site-packages/binwalk/core/module.py` exist?

closed 2016-11-22 by devttys0

blackt3a, 2024-07-01:
sudo apt purge binwalk
sudo apt autoclean
sudo apt-get install binwalk
python3 -m pip install git+https://github.com/ReFirmLabs/binwalk