ReFirmLabs/binwalk #212: AttributeError: 'module' object has no attribute 'MAGIC_NO_CHECK_TEXT'

DeadNumbers, 2016-05-28:
Hi! 
Ubuntu 12.04, Python 2.7.6
>sudo pip install git+https://github.com/devttys0/binwalk
>sudo pip install git+https://github.com/ahupp/python-magic
>$ binwalk
```
Traceback (most recent call last):
  File "/usr/bin/binwalk", line 5, in <module>
    import binwalk
  File "/usr/lib/python2.7/dist-packages/binwalk/__init__.py", line 12, in <module>
    class Binwalk:
  File "/usr/lib/python2.7/dist-packages/binwalk/__init__.py", line 28, in Binwalk
    DEFAULT_FLAGS = magic.MAGIC_NO_CHECK_TEXT | magic.MAGIC_NO_CHECK_ENCODING | magic.MAGIC_NO_CHECK_APPTYPE | magic.MAGIC_NO_CHECK_TOKENS
AttributeError: 'module' object has no attribute 'MAGIC_NO_CHECK_TEXT'
```

What i do wrong?

Also, install from repo binwalk and python-magic, but this error repeat.

devttys0, 2016-06-03:
`https://github.com/ahupp/python-magic` is not the python-magic you are looking for. It is incompatible with the python module actually distributed *with* libmagic. I suggest you remove it and install python-magic via apt-get: `sudo apt-get install python-magic`.

However, that is irrelevant for binwalk, because binwalk does not rely on python-magic, and has not for some time. It sounds like you have an old version of binwalk installed on your system along side the new one you just installed. Remove the old version completely - including the python module - and you should no longer get that error.

closed 2016-06-26 by DeadNumbers

maxi1192, 2016-11-04:
actually I got the same error message but I installed python-magic using apt-get. The root cause must be somewhere else.